raspberrypi/piserver #99: Occasional error in login with mounting home directory

rprr, 2020-03-23:
Sometimes the login of one of the users to the netbooted Pi seems to partially work but cannot get to the home directory. A second user seems to be able to login just fine. I see the following mount related error in /var/log/auth.log. Not using ldap. 
```
Mar 22 20:52:17 netpi sshd[29827]: pam_unix(sshd:session): session opened for user netuser by (uid=0)
Mar 22 20:52:17 netpi sshd[29827]: (mount.c:72): Messages from underlying mount program:
Mar 22 20:52:17 netpi sshd[29827]: (mount.c:76): fusermount: user has no write access to mountpoint /home/netuser
Mar 22 20:52:17 netpi sshd[29827]: (pam_mount.c:522): mount of netuser@piserver: failed
Mar 22 20:52:17 netpi systemd-logind[273]: New session c62 of user netuser.
```
Eventually, it seems to clear by itself and the user can log back into the Pi. How does one debug this?

maxnet, 2020-03-23:
Is there any pattern in which users it affects?
E.g. does it only happen to users that login for the first time, or is it more random?
(On first login the home directory should be created on the server by pam_mkdir)

>Not using ldap.

I am assuming you mean you are not using an EXTERNAL LDAP server.
And that the user does was created through the Piserver GUI, which creates the user on the LDAP server running on the Piserver computer.
(Just making sure you did not create the user through other means, like adduser).

rprr, 2020-03-24:
@maxnet -- thanks.

Yes, not EXTERNAL LDAP.

User was created through the Piserver GUI.

It appears to be more random. I think it seems to happen after the user has been logged for some time (say an hour) and been running various programs. Lots of python + hardware etc.

This particular user has elevated privileges (sudo, can reboot the pi, start up services etc) and also belongs to group that can control i2c, spi, gpio etc.

This hasn't happened all of today but logins today have been minimal. The Pi mostly runs programs that get started on boot via systemd.