raspberrypi/piserver #102: Automatic add new pi

m-gavrilyuk, 2020-04-11:
Is it possible to auto detect and add as a client new raspberry in network.
For example in some VLAN all raspberry was add as a client without GUI adding.

maxnet, 2020-04-11:
Only if it is okay that ALL Pi are going to run same OS with no exceptions.

Requires that you add stuff to `/etc/dnsmasq.conf` to tag all Pi to piserver group, among the lines of:

```
dhcp-mac=set:piserver,B8:27:EB:*:*:*
dhcp-mac=set:piserver,DC:A6:32:*:*:*
```

And symlink tftproot to the boot folder of the OS you want to use for all.

```
cd /var/lib/piserver
sudo mv tftproot tftproot.old
sudo ln -sf os/name-of-os/boot tftproot 
```

m-gavrilyuk, 2020-04-11:
Thanks for quick answer. Maybe add some wiki for questions for dont open new issue every time ?) Thanks

closed 2020-04-11 by m-gavrilyuk