raspberrypi/piserver #103: Dont see pi in wizard setup

m-gavrilyuk, 2020-04-17:
Hello can you tell what can be problem?
which ports should be open? 
On screen you can see that 1 device connect to lan. 

[image]

maxnet, 2020-04-17:
ports: UDP 67/68

But usually the problem -in case you are using a managed Ethernet switch- is that STP is enabled.
Disable it, or at least set it to the portfast (Cisco) or RSTP flavour.

m-gavrilyuk, 2020-04-18:
We use cisco 2960 switch and rstp is enable. 
Portfast not in option. Too many pis(100+)
It was hard to administrate.
Or i can only do it on server port ?

m-gavrilyuk, 2020-04-18:
how does stp  affect on piserver?

maxnet, 2020-04-18:
>how does stp affect on piserver?

With STP it takes 30+ seconds after the network link goes up before the network port actually works and forwards packets.
In that time it is testing for network loops.

Pi 3 gives up network booting before that time is over.

m-gavrilyuk, 2020-04-19:
is there any other way around this?
is there a delay function or increase the wait time for netboot?

maxnet, 2020-04-19:
With Pi 4 there is.
Judging on the MAC-address you are using Pi 3 though. That is not going to work without SD cards, if you use gear that has STP on by default (e.g. Cisco).

m-gavrilyuk, 2020-04-25:
PVST is work fine. up ling 2-3 sec 
Problem was that main dhcp server was in different Vlan so need to configure Relay on L3 cisco 
for piserver 
Thanks for help !!!

closed 2020-04-25 by m-gavrilyuk

m-gavrilyuk, 2020-04-26:
@maxnet 
The problem with managed switch is still available.
When reboot or power on client Pi after link come up nothing will happen
But if i disconnect ethernet cable and connect it again client will boot.
Portfast is active on interface.
Can piserver work with managed switch in case ? Or its only for unmanaged lan device ?

maxnet, 2020-04-26:
> Portfast is active on interface.

If you run tcpdump/wireshark on a Pi/computer attached to that exact port, and look at the stp packets that are being broadcasted, does that confirm forward-delay is 0?

m-gavrilyuk, 2020-04-26:
OK, in order to make this work for us, May I kindly ask U to show me the typical config of the ordinary access-port of Cisco switch?

maxnet, 2020-04-26:
>OK, in order to make this work for us, May I kindly ask U to show me the typical config of the ordinary 
>access-port of Cisco switch?

Sorry, but I am in the Juniper camp.
Don't have a Cisco handy right now.

m-gavrilyuk, 2020-04-26:
OK, then please the config of access-port of ordirary Juniper switch.

maxnet, 2020-04-26:
>OK, then please the config of access-port of ordirary Juniper switch.

They come with STP disabled by default, so are fine out-of-the-box.
STP really only has a place on Ethernet switches connecting other switches to each other, not for switches that connect computers.

Typically config of one of my switches (L2, in this particular case I have assigned half of the ports to a different VLAN, but you would probably not do that if using all for piserver):

```
system {
    host-name ex4200;
    time-zone Europe/Amsterdam;
    root-authentication {
        encrypted-password "*MY PASSWORD HASH*";
    }
    services {
        ssh {
            protocol-version v2;
        }
        netconf {
            ssh;
        }
        web-management {
            http;
            https {
                system-generated-certificate;
            }
        }
    }
    syslog {
        user * {
            any emergency;
        }
        file messages {
            any notice;
            authorization info;
        }
        file interactive-commands {
            interactive-commands any;
        }
    }
}
chassis {
    alarm {
        management-ethernet {
            link-down ignore;
        }
    }
}
interfaces {
    ge-0/0/0 {
        unit 0 {
            family ethernet-switching;
        }
    }
    ge-0/0/1 {
        unit 0 {
            family ethernet-switching;
        }
    }
    ge-0/0/2 {
        unit 0 {
            family ethernet-switching;
        }
    }
    ge-0/0/3 {
        unit 0 {
            family ethernet-switching;
        }
    }
    ge-0/0/4 {
[... 286 lines cut ...]
            ge-0/0/29.0;
            ge-0/0/30.0;
            ge-0/0/31.0;
            ge-0/0/32.0;
            ge-0/0/33.0;
            ge-0/0/34.0;
            ge-0/0/35.0;
            ge-0/0/36.0;
            ge-0/0/37.0;
            ge-0/0/38.0;
            ge-0/0/39.0;
            ge-0/0/40.0;
            ge-0/0/41.0;
            ge-0/0/42.0;
            ge-0/0/43.0;
            ge-0/0/44.0;
            ge-0/0/45.0;
            ge-0/0/46.0;
            ge-0/0/47.0;
        }
    }
}
```

(while it says "protocol rstp;" it is not actually active, as there are no interfaces/vlans assigned to it)

m-gavrilyuk, 2020-04-27:
For now i can use cd card with bootcode.bin
 but @ghollingworth  is there some hope to fix that bug in next "brunch" firmware?
Maybe possible to do some delay in initialisation ethernet port for pi 3 ?
Thanks

maxnet, 2020-04-27:
>Maybe possible to do some delay in initialisation ethernet port for pi 3 ?

Keep in mind that the bootrom on Pi 3 is not flashable, so would need new chip.
If you would need to buy new hardware anyway, why not upgrade to Pi 4?

m-gavrilyuk, 2020-04-27:
It will take long time for explain my production why we need to by new 100+ PI 4 to replace almost new Pi 3 ))) 
I found in documentation 
program_usb_boot_timeout=1 
Can it help ?

maxnet, 2020-04-27:
>It will take long time for explain my production why we need to by new 100+ PI 4 to replace almost new Pi 3 )))

The bootrom inside your Pi 3 is read-only memory.
So there is no way to get new firmware inside there.

Can only keep SD card with bootcode.bin in there permanently, upgrade to Pi 4 or fix your switch config.

m-gavrilyuk, 2020-04-27:
Thanks for help !