raspberrypi/piserver #104: link rootfs

m-gavrilyuk, 2020-04-30:
please can U help me, if I have one system from which I will make a clone, can I make links to the same directories from the clone in the original?
I have autologon configured and I want to have a lot of different ones when choosing a system for clients via the GUI, but in fact these would all be links to 1 rootfs with the exception of only some folders that would remain unique for each.
P.S Threat links to save server disk space
Sorry if some wrongs, i write in translate )

m-gavrilyuk, 2020-04-30:
I try do symbolic link (example /usr/lib/ 2GB) but when boot clients i have kernel panic

maxnet, 2020-04-30:
Symbolic links will not work.
Hard links may.

But I think you are making things more complicated than the problem you are trying to solve.

m-gavrilyuk, 2020-04-30:
I want autologon but with different folder for all. (100+ clints )

maxnet, 2020-04-30:
Still don't think it is a good solution.
Are you going to keep 100 OS instances up-to-date?

But if you really want to go that route, use hard links.

```
sudo cp -al original-folder copy-of-folder
```

Verify that it made hard links by doing "stat somefile", "links" should say 2.

m-gavrilyuk, 2020-05-01:
Thanks it work! 
I don't think i need to update every time
Its for one time preference and forgot.

closed 2020-05-01 by m-gavrilyuk