raspberrypi/piserver #105: Server and clients in different Vlans

m-gavrilyuk, 2020-05-04:
Pi server in 192.168.1.200
Real dhcp server in 192.168.1.20,  192.168.1.21 ( 2 server on windows server) 
Vlan for clients are 192.168.9.0/24 
In L3 router configure ip helper address 192.168.1.200  ip helper address 192.168.1.20  ip helper address 192.168.1.21
If i connect pi in 192.168.1.0 its work fine 
but in 192.168.9.0 i see only this 

vendor class: PXEClient:Arch:00000:UNDI:002001
May  4 11:25:31 Pi-server dnsmasq-dhcp[666]: 653460281 PXE(eth0) b8:27:eb:6a:38:59 proxy
May  4 11:25:31 Pi-server dnsmasq-dhcp[666]: 653460281 tags: piserver, known, client_is_a_pi, eth0
May  4 11:25:31 Pi-server dnsmasq-dhcp[666]: 653460281 reply delay: 2
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 broadcast response
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size:  1 option: 53 message-type  2
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size:  4 option: 54 server-identifier  10.121.0.236
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size: 17 option: 97 client-machine-id  00:59:38:6a:19:59:38:6a:19:59:38:6a:19:59...
May  4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
May  4 11:25:59 Pi-server dnsmasq-dhcp[666]: 980393422 available DHCP subnet: 192.168.1.200/255.0.0.0

and nothing more.
What i miss?

maxnet, 2020-05-04:
>Pi server in 192.168.1.200 
	May 4 11:25:33 Pi-server dnsmasq-dhcp[666]: 653460281 sent size: 4 option: 54 server-identifier 10.121.0.236 
More than one NIC and the other has 10.121.0.236 ?

m-gavrilyuk, 2020-05-04:
its for example 192.168.1.200

maxnet, 2020-05-04:
>its for example 192.168.1.200

And your real network has /16 subnets?
(If not I would expect the netmask to be 255.255.255.0 on the Piserver server...)

maxnet, 2020-05-04:
>May 4 11:25:59 Pi-server dnsmasq-dhcp[666]: 980393422 available DHCP subnet: 
>192.168.1.200/255.0.0.0

You would only use netmask 255.0.0.0 if you do not have any subnets inside your 10.x.x.x network.

m-gavrilyuk, 2020-05-04:
Sory my bad i dont see this 
Now its /24 mask 
and i see no address range available for DHCP request via 192.168.9.2 ( its ip of vlan interface)

maxnet, 2020-05-04:
May need to add the foreign ranges manually to `/etc/dnsmasq.conf`

```
dhcp-range=192.168.9.0,proxy,255.255.255.0
```

m-gavrilyuk, 2020-05-04:
available DHCP subnet: 192.168.9.0/255.255.255.0
May  4 12:34:33 Pi-server dnsmasq-dhcp[21599]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
May  4 12:34:33 Pi-server dnsmasq-dhcp[21599]: 653460281 PXE(eth0) b8:27:eb:6a:38:59 proxy
May  4 12:34:33 Pi-server dnsmasq-dhcp[21599]: 653460281 tags: piserver, known, client_is_a_pi, eth0
May  4 12:34:33 Pi-server dnsmasq-dhcp[21599]: 653460281 reply delay: 2
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 broadcast response
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 sent size:  1 option: 53 message-type  2
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 sent size:  4 option: 54 server-identifier  192.168.1.200
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 sent size: 17 option: 97 client-machine-id  00:59:38:6a:19:59:38:6a:19:59:38:6a:19:59...
May  4 12:34:35 Pi-server dnsmasq-dhcp[21599]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...

maxnet, 2020-05-04:
If it is an orignal Pi 3 (not +), try if it works better if you insert SD card with bootcode.bin

m-gavrilyuk, 2020-05-04:
> If it is an orignal Pi 3 (not +), try if it works better if you insert SD card with bootcode.bin
Its already  with it

m-gavrilyuk, 2020-05-04:
maybe i need trunk to piserver and manually connect sub interface to 192.168.9.0/24 ?

maxnet, 2020-05-04:
>maybe i need trunk to piserver and manually connect sub interface to 192.168.9.0/24 ?

While that obviously should work as well, it should not be necessary.

And from the looks of it your DHCP relaying does work fine. Just not much happening afterwards.
The routing of your L3 router does work correctly?
If -for testing purposes- you put a SD card with Raspbian in it in a Pi, and give it a static 192.168.2.x/24 IP, you can ping the piserver PI and vice versa?

m-gavrilyuk, 2020-05-04:
i can do it from all other device in different subnets and in 192.168.9.0/24 too

maxnet, 2020-05-04:
If you have a serial console cable laying around you can also try if enabling the debugging in bootcode.bin reveals anything: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/README.md

m-gavrilyuk, 2020-05-04:
I can see only Unicode in terminal console

maxnet, 2020-05-04:
When using serial console?
Make sure you have right baud rate (115200)

m-gavrilyuk, 2020-05-04:
I connect to another pi pin to pin (cross-connect)
Screen /dev/ttyS0 115200
And see lot of unicode symbols

maxnet, 2020-05-04:
If you really want to use a Pi I suggest putting enable_uart=1 in config.txt and using /dev/ttyAMA0
But strongly suggest a normal USB uart cable. Less that can go wrong with that.

m-gavrilyuk, 2020-05-04:
I replace my server to 192.168.9.0 subnet 
Its help client now boot, but they have link-local address 169.254.x.x
But ifconfig show me 192.168.9.x 
If try ping anything it do it from 169.254.x.x address
I dont understand whats happend.

maxnet, 2020-05-04:
>Its help client now boot, but they have link-local address 169.254.x.x
>But ifconfig show me 192.168.9.x

Sorry, not following you here.
Where does it say they have a link-local address?

m-gavrilyuk, 2020-05-04:
When ping anything he show ping from and this link-local address

maxnet, 2020-05-04:
You do are assigning it a valid gateway address (one that is inside the same subnet as the Pi) through DHCP?

"ip route" confirms a gateway is set?

m-gavrilyuk, 2020-05-04:
Route show me default 192.168.9.0 
And second gw link-local 
If i add static route by my self it work

maxnet, 2020-05-04:
>Route show me default 192.168.9.0

You cannot use the network IP (.0) for gateway.
First usable IP of a /24 subnet is .1

m-gavrilyuk, 2020-05-04:
Route 
Destination.   Gateway.   Genmask
Default           0.0.0.0          0.0.0.0
192.168.9.0   0.0.0.0.        255.255.255.0
Link-local.      0.0.0.0         255.255.0.0

maxnet, 2020-05-04:
>Default 0.0.0.0 0.0.0.0

Your missing a default gateway...

m-gavrilyuk, 2020-05-04:
But others machine's windows and Linux
In this subnet work fine with all right dhcp options.

m-gavrilyuk, 2020-05-07:
https://github.com/Lukanite/PiServer-PXE-Serial-Translator/wiki

I see the patch for dnsmasq is it can help me?

maxnet, 2020-05-07:
You will probably need that as well, but it is not your current problem.

m-gavrilyuk, 2020-05-07:
Maybe try standalone dhcp server role in pi server ?

maxnet, 2020-05-07:
Only do so if there is not any other DHCP server active in the VLAN piserver is in.
I don't know anything about your network architecture, so cannot judge if that is a good idea...

m-gavrilyuk, 2020-05-09:
the problem was solved using / network / interfaces auto eth0 iface eth0 inet dhcp what turns off dhcpcd and all network settings began to arrive right but the server is in the same network segment as the clients

maxnet, 2020-05-09:
dhcpcd may be more picky then the dhcp client you get when using dhcp through /etc/network/interfaces.
Would still investigate why dhcpcd does not like the lease it gets from your dhcp server though, instead of blaming dhcpcd.

Gateway with an IP outside usable IP subnet range?

m-gavrilyuk, 2020-05-09:
I did it by accident and it helped. I specially created a primitive network with two vlanes and a dhcp server from scratch and the result was the same as in the working network of the enterprise so there’s nothing more to blame

closed 2020-09-08 by m-gavrilyuk