raspberrypi/piserver #107: access management

m-gavrilyuk, 2020-05-27:
Is it possible to access only to add new client for some users in piserver ?

maxnet, 2020-05-27:
Not sure if I understand the question.

By default you need to be a user that is part of the sudo group to access the Piserver GUI.
Or is that not what you are asking?

m-gavrilyuk, 2020-05-27:
No. I want add local users but only with right to add new client on piserver.
Other functions must be disabled

maxnet, 2020-05-27:
You want to let other staff login locally to the server that runs Piserver, and allow them to only add extra Pi MAC addresses to the system, but do not let them do anything else?

m-gavrilyuk, 2020-05-27:
Yes. there are no other ideas for delimiting rights

maxnet, 2020-05-27:
That is not something we support.

Not sure if I understand the exact need for it either.
I would have expected that adding new Pi to the system is something one would do a lot less often, than say reset the password of an user that forget his.

m-gavrilyuk, 2020-05-27:
I use piserver for information display in kiosk mode so they all with autologon and no need to manage users 
Thanks i will other staff dont crash server.

closed 2020-05-27 by m-gavrilyuk

m-gavrilyuk, 2020-05-28:
Can you tell me please. how users can run piserver without sudo privileges?

maxnet, 2020-05-28:
>Can you tell me please. how users can run piserver without sudo privileges?

Not possible in a secure manner.
(In sudo configuration you do can specify that users are only allowed to start piserver with "sudo" and no other commands.
But they will effectively still have root access then, through the chroot shell button in Piserver. Chroot does not provide any real protection)

You will need to create a webinterface or other abstraction layer if you want to allow users to perform only a limited number of privileged operations.
Outside the scope of piserver.