raspberrypi/piserver #112: Hangs on boot to GUI

OkenKhuman, 2020-10-22:
When I boot to console there is no problem but when I try booting into GUI desktop client RPi got hang.

I enabled the GUI boot by:
installing `lightdm`
enabling boot option to Desktop

But on booting it got stuck on rainbow splash screen.

What can I do to resolve this?

maxnet, 2020-10-22:
I am guessing you are not using the "Raspberry Pi OS" that is provided by Piserver as base, as that already has lightdm by default. So what Linux distribution are you using instead? RPI OS Lite? Something else?

The modifications we normally do to make RPI OS suitable for read-only environments are listed here: https://github.com/raspberrypi/piserver/blob/master/scripts/convert_chrooted_cmds
May need to do those affecting lightdm manually. Like:

- add "tmpfs           /var/lib/lightdm  tmpfs defaults,mode=755,nosuid     0       0" to /etc/fstab manually.
- set user-authority-in-system-dir=true in /etc/lightdm/lightdm.conf

OkenKhuman, 2020-10-22:
Thanks for the speedy reply.
I am using the "Raspbian Lite (version 2020-02-13)".

I will try out the the changes.

OkenKhuman, 2020-10-22:
[image]

Seems like lightdm problem.

maxnet, 2020-10-22:
Also make sure you have autologin disabled (unless you mounted /home/pi as tmpfs accordingly)

OkenKhuman, 2020-10-22:
This GUI boot problem was solved with `Raspbian` image. No more GUI boot problem found. 
Thanks for help.

closed 2020-10-22 by OkenKhuman