raspberrypi/piserver #113: Problem in detecting clients for piserver installed on RPi4

OkenKhuman, 2020-10-24:
I successfully compiled, install and run the piserver on RPi4 but I am not able to detect any client...

I was able to detect client on VM based piserver

What can I do?

maxnet, 2020-10-24:
Compiled under RPI OS or a different Linux distribution for the Pi 4?

Your Pi 4 and the Pi being network booted are both connected to the exact same Ethernet switch you were using under the VM based setup? And they live in the same VLAN?

OkenKhuman, 2020-10-24:
My RPi4 (hosting the piserver) is having `raspbian buster` kernel is `Linux RPi4 5.4.70-v7l-custom+ #2 SMP Fri Oct 16 19:45:03 IST 2020 armv7l GNU/Linux` 
Same network setup as the VM setup (which works)
Same VLAN
The RPi3b I am trying to network boot has already been network booted from the VM setup.

Only problem:
Cannot be detected by my piserver running on RPi4

maxnet, 2020-10-24:
And the Pi 3B being network booted is connected to the exact same network port as with the VM setup previously, right?
Double checking, because on managed Ethernet switches settings like STP (which can cause trouble with network boot) are configurable by port.

There are no security settings like DHCP snooping active on the switch? (If it is a managed one).

And you have no iptables/nft rules on the Pi?

OkenKhuman, 2020-10-24:
No different network port.
My switch is a semi-managed `tl sg105e` so no much option as DHCP security I guess.

When I nmap the network (turning off the client and starting both piserver)
```
pi@raspberry:~ $ nmap 192.168.0.*
Starting Nmap 7.70 ( https://nmap.org ) at 2020-10-24 14:34 BST
Nmap scan report for 192.168.0.1
Host is up (0.015s latency).
Not shown: 999 filtered ports
PORT   STATE SERVICE
80/tcp open  http

Nmap scan report for piserver (192.168.0.2)
Host is up (0.000019s latency).
Not shown: 995 closed ports
PORT     STATE SERVICE
53/tcp   open  domain
111/tcp  open  rpcbind
389/tcp  open  ldap
1022/tcp open  exp2
2049/tcp open  nfs

Nmap scan report for 192.168.0.3
Host is up (0.00056s latency).
Not shown: 991 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
53/tcp   open  domain
111/tcp  open  rpcbind
139/tcp  open  netbios-ssn
389/tcp  open  ldap
445/tcp  open  microsoft-ds
1022/tcp open  exp2
2049/tcp open  nfs
5900/tcp open  vnc

Nmap done: 256 IP addresses (3 hosts up) scanned in 9.25 seconds
```
And I used different port with different static IP for the RPi4 piserver and VM piserver

```
		RPi4 piserver	VM piserver
IP		192.168.0.3		192.168.0.2
Port	        2				1

```
while I plug my client RPi3b on port 5.

maxnet, 2020-10-24:
Suggests you try a clean RPI OS first.
Without any additional custom stuff (see you at least have docker and samba installed).

Piserver should be available through apt. Shouldn't be necessary to compile it yourself either.

OkenKhuman, 2020-10-24:
Looks like somewhat hard (because my RPi4 setup is headless one) to do but I will try experimenting week

I will update here if I find any difficulty or if I succeed

OkenKhuman, 2020-10-24:
Finally I am able to network boot my RPi3b from RPi4 (piserver).
Thanks for guiding me.

It happens that there were some bugs on local build, so by installing the pre-build package and updating the network configurations it works.

No more problem found.

closed 2020-10-24 by OkenKhuman