raspberrypi/piserver #114: Docker installation to served image

OkenKhuman, 2020-10-25:
I am finding difficulty since the client user is omitted to use sudo permission and docker need sudo permission to initialize the setup. (i.e. to add user and proceed with the docker operations)

I sucessfully installed the docker on image(software) by using shell inbuild but the problem is:
*no permission to connect to daemon socket

Will there be possible to provide sudo previledge to client (selectively)?

maxnet, 2020-10-25:
Nothing prevents you from allowing certain users to run certain commands as root in `/etc/sudoers`

That is not going to allow you to add users though.
The root file system is read-only, nobody can write to it on the client, not even root.

OkenKhuman, 2020-10-25:
[image]

OkenKhuman, 2020-10-25:
I think I fund the answer , will try it 
[https://github.com/raspberrypi/piserver/issues/5](url)

maxnet, 2020-10-25:
>I think I fund the answer

No, you did not.
Enabling sudo by itself will not allow usermod to write to /etc/group on the client.

If you want to add a specific user to a group, and you know the username of the user in advance, open a chroot shell in Piserver on the server, and edit the group file manually ("leafpad /etc/group") to add that user to the group you want.

If it is not for a single named user, but all Piserver network users are supposed to be member of the group "docker", you can use pam_group to accomplish that instead.
On the Piserver server, open a chroot shell, "leafpad /etc/security/group.conf"
Add "docker" to the list on the line mentioning lightdm at the end of the file.
That will only work for users that are logging in interactively through the GUI though.

OkenKhuman, 2020-10-25:
Thanks I did make some progress by this mods. I am now able to do previledged work on the client up to some extent and also give docker commands.

But is there a way to make some root directory (like /var and /tmp) permit read/write/execute  because I am not able to run docker daemon.

OkenKhuman, 2020-10-25:
[image]

maxnet, 2020-10-25:
>But is there a way to make some root directory (like /var and /tmp) permit read/write/execute 
>because I am not able to run docker daemon.

Not when you plan to share the same file system with more than one Pi.

Naturally, if you have a separate operating system installation for each Pi you want to network boot, you do can make the NFS share read-write.
Do be aware of the security implications though. It is only safe to use read-write NFS with IP based authenticated in a private network your control.
Fine at home. Not so suitable for corporate or school networks.

OkenKhuman, 2020-10-25:
I can make my server store seperate image for each client. Where can I modify the image to make it read-write accessable especially /var?

maxnet, 2020-10-25:
On the server change "ro" for "rw" in /etc/exports, and restart nfs server (or entire server), and add " rw" to the same line as existing options in `/var/lib/piserver/os/name-of-os/boot/cmdline.txt`

OkenKhuman, 2020-10-25:
I see this line `console=serial0,115200 console=tty1 elevator=deadline rootwait quiet splash plymouth.ignore-serial-consoles ip=dhcp root=/dev/nfs nfsroot=192.168.0.3:/var/lib/piserver/os/Raspbian_RPi3b,v3` on `cmdline.txt` 

Should I put `,rw` or `rw` at the end of line?

maxnet, 2020-10-25:
In cmdline.txt " rw"
Separated by space from previous option, no comma.

OkenKhuman, 2020-10-25:
Thanks again. Now the read-write works on root areas.

closed 2020-10-25 by OkenKhuman