raspberrypi/piserver #116: Enabling WiFi on client with read, write and root permission.

OkenKhuman, 2020-11-08:
My RPi3b client (with root, read and write permission) cannot detect it's build in WiFi chip. 
`No network interface was found` was shown when I try to enable WiFi via `raspi-config`.
How to proceed?

maxnet, 2020-11-08:
From the chroot shell in the Piserver GUI:
1) `leafpad /boot/config.txt` - remove the `dtoverlay=pi3-disable-wifi` line
2) If you want to use wifi as client (as opposed as playing access point for others) `leafpad /etc/dhcpcd.conf` - remove `nolink`

Removing "nolink" does have the side effect that if your LAN connection is unplugged your system will hang, and it will not reconnect even if you plug it in again.
(dhcpcd wants to execute scripts on link change, and that is hard if the network file system is no longer available).

OkenKhuman, 2020-11-08:
It works.

But is it possible to share internet connection from the client to the piserver hosting RPi?

maxnet, 2020-11-08:
> But is it possible to share internet connection from the client to the piserver hosting RPi?

Only if the client has a fixed IP known in advance.

Is Piserver's DHCP server handing out the IPs, or is another DHCP server doing that and Piserver running proxydhcp?

OkenKhuman, 2020-11-08:
I think my piserver is handeling the DHCP work in my setup because there is only : semi- managed switch[10.101.0.1] , RPi4(piserver)[10.101.0.3], RPi3b(client)[ip 10.101.0.46] and laptop on totally different ip-range

```
Starting Nmap 7.91 ( https://nmap.org ) at 2020-11-08 15:13 UTC
Nmap scan report for 10.101.0.1
Host is up (0.016s latency).
Not shown: 999 filtered ports
PORT   STATE SERVICE
80/tcp open  http

Nmap scan report for 10.101.0.3
Host is up (0.000046s latency).
Not shown: 991 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
53/tcp   open  domain
111/tcp  open  rpcbind
139/tcp  open  netbios-ssn
389/tcp  open  ldap
445/tcp  open  microsoft-ds
1022/tcp open  exp2
2049/tcp open  nfs
5900/tcp open  vnc

Nmap scan report for 10.101.0.46
Host is up (0.00051s latency).
Not shown: 999 closed ports
PORT   STATE SERVICE
22/tcp open  ssh

Nmap done: 256 IP addresses (3 hosts up) scanned in 37.98 seconds
```

maxnet, 2020-11-08:
>I think my piserver is handeling the DHCP work in my setup

What does it say it is in the "settings" tab of the Piserver GUI?

OkenKhuman, 2020-11-08:
Standalone DHCP server

maxnet, 2020-11-08:
You can probably assign a static IP to the client by putting its MAC address in /etc/ethers along the lines of:

```
11:22:33:44:55:66 10.101.0.123
```

Where 11:22:33:44:55:66 is the MAC of client.

You can then configure the static network configuration of the Piserver server to use 10.101.0.123 as gateway.
And setup iptables/npt rules on the client to share the wifi connection with the Piserver box.
E.g. like described here: https://wiki.archlinux.org/index.php/Internet_sharing#With_nftables
(use wlan0 where it says internet0)

OkenKhuman, 2020-11-08:
iptable is hard for me so I accomplished using locally run openvpn client-server setup.

closed 2020-11-08 by OkenKhuman