raspberrypi/piserver #117: Swap file issue

OkenKhuman, 2020-11-11:
when trying to upgrade I came across this issue
```
pit@raspberrypi:~ $ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libmicrodns0 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up dphys-swapfile (20100506-5+rpt1) ...
Job for dphys-swapfile.service failed because the control process exited with error code.
See "systemctl status dphys-swapfile.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dphys-swapfile, action "restart" failed.
● dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file
   Loaded: loaded (/lib/systemd/system/dphys-swapfile.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-11-11 12:55:33 IST; 53ms ago
     Docs: man:dphys-swapfile(8)
  Process: 31460 ExecStart=/sbin/dphys-swapfile setup (code=exited, status=1/FAILURE)
 Main PID: 31460 (code=exited, status=1/FAILURE)

Nov 11 12:55:32 raspberrypi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Nov 11 12:55:33 raspberrypi dphys-swapfile[31460]: want /var/swap=100MByte, checking existing: deleting wrong size file (0), generating swapfile ...
Nov 11 12:55:33 raspberrypi systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=1/FAILURE
Nov 11 12:55:33 raspberrypi systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Nov 11 12:55:33 raspberrypi systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
What is needed to change?

maxnet, 2020-11-11:
On the normal Piserver images swap is disabled (the equivalent of "systemctl disable dphys-swapfile").

Does your use-case really need it?

OkenKhuman, 2020-11-11:
Yes I think to fully operate Docker and come complex containers need it.

OkenKhuman, 2020-11-11:
One of the problem is 
```
pit@raspberrypi:~ $ sudo apt remove docker-ce
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'docker-ce' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  containerd.io docker-ce-cli libmicrodns0 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up dphys-swapfile (20100506-5+rpt1) ...
Job for dphys-swapfile.service failed because the control process exited with error code.
See "systemctl status dphys-swapfile.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dphys-swapfile, action "restart" failed.
● dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file
   Loaded: loaded (/lib/systemd/system/dphys-swapfile.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-11-11 20:18:59 IST; 49ms ago
     Docs: man:dphys-swapfile(8)
  Process: 1765 ExecStart=/sbin/dphys-swapfile setup (code=exited, status=1/FAILURE)
 Main PID: 1765 (code=exited, status=1/FAILURE)

Nov 11 20:18:58 raspberrypi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Nov 11 20:18:59 raspberrypi dphys-swapfile[1765]: want /var/swap=100MByte, checking existing: deleting wrong size file (0), generating swapfile ...
Nov 11 20:18:59 raspberrypi systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=1/FAILURE
Nov 11 20:18:59 raspberrypi systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Nov 11 20:18:59 raspberrypi systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

maxnet, 2020-11-11:
I am personally not a big fan of swap.
Both from the performance perspective and the security one.
(As an user I do not like the concept that if I type confidential information on my computer -such as passwords and credit card numbers- there is a risk those are swapped from memory to disk.)

That said, if you really want to have swap:
I think the kernel does have the necessary options to have a swap file on NFS, but the problem is that RPi OS is using a non-standard dphysfile startup script that wants to create the swap file sparse, which is not supported on NFS.

Try if things work better if you replace in /sbin/dphys-swapfile :

```
      if type fallocate > /dev/null && [ "${FS_TYPE}" != "f2fs" ] ; then
         fallocate -l "${CONF_SWAPSIZE}"M "${CONF_SWAPFILE}" 2> /dev/null
       else
         dd if=/dev/zero of="${CONF_SWAPFILE}" bs=1048576 \
```

With:

```
      if type fallocate > /dev/null && [ "${FS_TYPE}" != "f2fs" ] && [ "${FS_TYPE}" != "nfs" ] ; then
         fallocate -l "${CONF_SWAPSIZE}"M "${CONF_SWAPFILE}" 2> /dev/null
       else
         dd if=/dev/zero of="${CONF_SWAPFILE}" bs=1048576 \
```

OkenKhuman, 2020-11-11:
yes replacing with the below script blocks works

Thank you

closed 2020-11-11 by OkenKhuman