raspberrypi/piserver #118: get login msg on piserver

chengjianwen, 2020-11-16:
How can I get a message of login? for example login name, login address, login time, and logout messages.
thus I can execute vncviewer to the special login by it's IP address.
I installed vnc server on the os, and enable it already.

I think there should be an pam_execute alike module to execute a command like hostnamectl set-hostname to set the pi's hostname to user's login name, and through mDNS to find the pi's IP address through the login name, but failed.

I also searched any method to execute script when a ssh client login (piserver use sshfs to mount user's home dir), but failed again.

any idea?

I kown Epoptes, but it not a solution for me.

maxnet, 2020-11-16:
>thus I can execute vncviewer to the special login by it's IP address.

Special login as in you only want to monitor specific users?!
Not sure if that is the kind of thing we should support...

chengjianwen, 2020-11-17:
I just want to set client pi a special host name to VNC connect from piserver.

closed 2020-11-19 by chengjianwen