raspberrypi/piserver #119: An overlayroot script worked fine for os

chengjianwen, 2020-11-19:
I installed the [https://github.com/chesty/overlayroot](overlayroot) script to piserver's os, and thus I can delete all tmpfs entries from /etc/fstab. It worked fine and thus the client can get an writable system temporary.

But it don't worked correctly with an new os converted from latest pi os with kernel 5.4 on init-bottom-overlay line 64: mount -t overlay overlay -olowerdir=/overlay/lower,upperdir=/overlay/upper,workdir=/overlay/work ${rootmnt} 
and with an error code: no such device.

maxnet, 2020-11-19:
You did recreate the initramfs on a system with the exact same kernel version used?
Initramfs contains kernel modules that must be exact same version as the running kernel.

chengjianwen, 2020-11-20:
Of course I did it that way. I created the converted os on an same os using a p4.