raspberrypi/piserver #120: support for: loading to RAM and booting from it

OkenKhuman, 2020-12-11:
How can I prepare to make sure the image is loaded to RAM and booted from it.

My RPi4 is with 4GB and since booting from network does reduce the performance, I thought boot from RAM will help me a lot.

My target is to:
1) Mount root FS on RAM. (so that client will not change the actual image and fast in file system)
2) Mount /home (no change to piserver style).
3) Finally to make client have all privilege.

OkenKhuman, 2020-12-15:
Sorry this is already there in the project.

I just have to modify some in image creation.

closed 2020-12-15 by OkenKhuman

m-gavrilyuk, 2021-02-25:
> Sorry this is already there in the project.
> 
> I just have to modify some in image creation.

Is this support only for PI4 ? 
when my pi 3 lost lan connection they just freeze and nothing work until connection will be restore

OkenKhuman, 2021-02-28:
@m-gavrilyuk I think RPi3 is not a good option to be a client of the setup. 
But if you go for non graphic image it works fine for both Pi3 and Pi4.