raspberrypi/piserver #121: How to enable VNC?

OkenKhuman, 2020-12-19:
How to enable VNC on client?

m-gavrilyuk, 2021-01-15:
raspi-config 
interfaces settings enable vnc ?

ilium007, 2021-03-02:
Can this be done by script ie. without having to interactively access raspi-config

maxnet, 2021-03-04:
>Can this be done by script ie. without having to interactively access raspi-config

You should only have to do it once in the chroot shell on the server, to have it apply to all clients.
So no need to script.

Do keep in mind that using RealVNC on more than 10 computers, or for commercial purposes may require commercial licensing.

hans-dot, 2024-02-12:
I did setup piserver out of the box from Debian Bullseye with Raspberry Pi Desktop found here:
https://www.raspberrypi.com/software/raspberry-pi-desktop/
I did try   raspi-config   to enable VNC server for the clients but gut error:
root@raspberry:/# raspi-config
cat: /proc/device-tree/model: No such file or directory
Running in chroot, ignoring command 'status'
Running in chroot, ignoring command 'start'
cat: /proc/device-tree/model: No such file or directory
any help?