raspberrypi/piserver #123: Sorry this is already there in the project.

m-gavrilyuk, 2021-02-28:
Sorry this is already there in the project.

I just have to modify some in image creation.

_Originally posted by @OkenKhuman in https://github.com/raspberrypi/piserver/issues/120#issuecomment-745240354_

Freeze all client when lan connection lost until it will be restore

maxnet, 2021-02-28:
>Sorry this is already there in the project.

Good to hear that it is already there.
Does that mean I can close this issue? :-)

Or are you meaning to ask: "is there are any way I can have my network booted operating system to continue running like nothing has happened, while my network connection is down"
That is only possible if everything you want to run is in RAM. Which is normally not the case.

Your average 4+ GB RPI OS Linux distribution is not going to fit comfortably there.
You do can build smaller (initramfs) operating system images that are so small they do fit with tools like: https://buildroot.org/
But that does require that you are know what you are doing. It is not a matter of setting a switch.

m-gavrilyuk, 2021-02-28:
Ok i got it ) its normal reaction. 
Is there some time out while connecting can be restore and client alive ?

closed 2021-02-28 by m-gavrilyuk

maxnet, 2021-02-28:
>Is there some time out while connecting can be restore and client alive ?

Recall NFS itself retries until the end of the days (with hard nfs mounts).
Although other factors may limit that.
If the client cannot reach the network's DHCP server either, its DHCP lease may expire.
And that may get you into a state you are unable to recover from.

(DHCP client is likely resident in RAM sufficiently enough, that it is able to remove the IP address from the network interface, as the client is no longer allowed to use it. But probably not capable of obtaining a new IP after connection comes back.)

m-gavrilyuk, 2021-03-01:
That is, if we ever switch to rpi 4 8 ​​gb ram, what i need to load FS in ram ?

maxnet, 2021-03-01:
>That is, if we ever switch to rpi 4 8 ​​gb ram, what i need to load FS in ram ?

Even if you have 8 GB RAM, you typically do still want to have a trimmed down file system image.
As even if you have the RAM, letting each client download a multiple gigabytes image on every boot is not going to speed things up...

If you want something based on RPI OS, the Debian Live helper stuff may suit your needs: https://live-team.pages.debian.net/live-manual/html/live-manual/index.en.html
Can use it to create a trimmed down image with a compressed filesystem.squahsfs image that can be downloaded to RAM in its entirety with the right options.

m-gavrilyuk, 2021-03-05:
My image of raspios 1gb if compress it with squash it can be less then 500 mb 
Maybe it can make possible to load fs to ram even for pi3 b + with 1GB ram ?

maxnet, 2021-03-05:
And you are running what kind of applications with that image?

To illustate: opening say Chrome and visiting the raspberrypi.org main page (single tab) already consumes 325 MB.

[code]
pi@raspberrypi:~ $ smem -t -P ^/usr/lib/chromium
  PID User     Command                         Swap      USS      PSS      RSS 
 1024 pi       /usr/lib/chromium-browser/c        0      180     1408    10412 
 1059 pi       /usr/lib/chromium-browser/c        0      932     4578    32772 
 1022 pi       /usr/lib/chromium-browser/c        0      304     4830    35484 
 1071 pi       /usr/lib/chromium-browser/c        0      880     5714    15904 
 1021 pi       /usr/lib/chromium-browser/c        0      632     6180    35836 
 1122 pi       /usr/lib/chromium-browser/c        0     2908     9972    52376 
 1050 pi       /usr/lib/chromium-browser/c        0    10356    17982    57616 
 1079 pi       /usr/lib/chromium-browser/c        0     4780    19123    79440 
 1045 pi       /usr/lib/chromium-browser/c        0    28772    47174    99152 
 1257 pi       /usr/lib/chromium-browser/c        0    35900    54545   120784 
  999 pi       /usr/lib/chromium-browser/c        0    57964    75615   138312 
 1080 pi       /usr/lib/chromium-browser/c        0    58260    78488   146956 
-------------------------------------------------------------------------------
   12 1                                           0   201868   325609   825044 
[/code]

Whenever the Linux kernel feels it is going to run low on memory (that is even before it is zero), it will start killing its offspring, starting by largest memory hog first.

So netbooting a 500 MB image when you only have 1 GB in total only makes sense when you are only starting applications that do not consume too much.

m-gavrilyuk, 2021-05-27:
What i need to change  on server to test load to RAM for 8GB pi4 ?

maxnet, 2021-05-27:
>@maxnet What i need to change on server to test load to RAM for 8GB pi4 ?

As mentioned before the Debian live helper stuff may do what you need.
Suggest you experiment a bit with that yourself, and come back if you have specific questions.

Getting such thing up & running *properly* is easily a day's work.
Writing a proper tutorial for this kind of thing would require a similar amount of time, and that is beyond the support I can give to you.