raspberrypi/piserver #125: not boot client

m-gavrilyuk, 2021-04-01:
hi help for debug i install new pi-server 
fresh os fresh chroot os 
ip-helper on L3 cisco is on 
i see client in wizard can add it choose os but no tftp load in logs
client and server in same Vlan

dnsmasq-dhcp[1321]: 653460281 available DHCP subnet: 10.20.35.254/255.255.255.0
Apr  1 13:04:59 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Apr  1 13:04:59 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 PXE(eth0) b8:27:eb:5e:2a:2e proxy
Apr  1 13:04:59 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 tags: piserver, eth0
Apr  1 13:04:59 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 reply delay: 1
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 broadcast response
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  1 option: 53 message-type  2
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  4 option: 54 server-identifier  10.20.35.254
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size: 17 option: 97 client-machine-id  00:2e:2a:5e:e7:2e:2a:5e:e7:2e:2a:5e:e7:2e...
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 available DHCP subnet: 10.20.35.254/255.255.255.0
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 PXE(eth0) b8:27:eb:5e:2a:2e proxy
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 tags: piserver, eth0
Apr  1 13:05:00 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 reply delay: 1
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 broadcast response
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  1 option: 53 message-type  2
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  4 option: 54 server-identifier  10.20.35.254
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size: 17 option: 97 client-machine-id  00:2e:2a:5e:e7:2e:2a:5e:e7:2e:2a:5e:e7:2e...
Apr  1 13:05:01 raspi-boot-srv dnsmasq-dhcp[1321]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...

and nothing more happend

maxnet, 2021-04-01:
>ip-helper on L3 cisco is on
>client and server in same Vlan

What is the use-case of DHCP forwarding (ip helper) if client and server live in same VLAN?

m-gavrilyuk, 2021-04-01:
Yes main dhcp in other vlan 
No dhcp snooping or acl in network

maxnet, 2021-04-01:
May want to setup traffic mirroring to another switch port, and run a packet capture there to verify that DHCP relaying is working properly, and the Pi does get a DHCP lease from your main DHCP server.

m-gavrilyuk, 2021-04-01:
is it possible than 66 option on my pool can disturb pi-server 
we have wds server setup in 66 option in main dhcp pool settings

maxnet, 2021-04-01:
>is it possible than 66 option on my pool can disturb pi-server

Judging on reports by others, yes.
https://www.raspberrypi.org/forums/viewtopic.php?p=1578751#p1578751

m-gavrilyuk, 2021-04-01:
ok i try disable this option 
i now for rpi 4 if manualy use TFTP_IP= in rpi-eeprom-config  it work fine with 66 option of deferent server
i will wire if help , thanks

m-gavrilyuk, 2021-04-02:
change option 66 
now i see tftp loading but client load again and again 

sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/start.elf to 10.20.35.130
Apr  2 13:58:54 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/fixup.dat to 10.20.35.130
Apr  2 13:58:54 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found
Apr  2 13:58:54 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:58:54 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/dt-blob.bin not found
Apr  2 13:58:55 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found
Apr  2 13:58:55 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 available DHCP subnet: 10.20.35.254/255.255.255.0
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 PXE(eth0) b8:27:eb:5e:2a:2e proxy
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 tags: known, piserver, eth0
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 reply delay: 1
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/e75e2a2e/start.elf not found
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/autoboot.txt not found
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:59:01 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 broadcast response
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  1 option: 53 message-type  2
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  4 option: 54 server-identifier  10.20.35.254
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size: 17 option: 97 client-machine-id  00:2e:2a:5e:e7:2e:2a:5e:e7:2e:2a:5e:e7:2e...
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 available DHCP subnet: 10.20.35.254/255.255.255.0
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 PXE(eth0) b8:27:eb:5e:2a:2e proxy
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 tags: known, piserver, eth0
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 reply delay: 1
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 broadcast response
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  1 option: 53 message-type  2
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  4 option: 54 server-identifier  10.20.35.254
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size: 17 option: 97 client-machine-id  00:2e:2a:5e:e7:2e:2a:5e:e7:2e:2a:5e:e7:2e...
Apr  2 13:59:03 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 sent size: 32 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Apr  2 13:59:04 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/start.elf to 10.20.35.130
Apr  2 13:59:04 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/fixup.dat to 10.20.35.130
Apr  2 13:59:04 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found
Apr  2 13:59:04 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:59:04 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/dt-blob.bin not found
Apr  2 13:59:05 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found
Apr  2 13:59:05 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:59:11 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 available DHCP subnet: 10.20.35.254/255.255.255.0
Apr  2 13:59:11 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Apr  2 13:59:11 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 PXE(eth0) b8:27:eb:5e:2a:2e proxy
Apr  2 13:59:11 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 tags: known, piserver, eth0
Apr  2 13:59:11 raspi-boot-srv-aaq dnsmasq-dhcp[31943]: 653460281 reply delay: 1
[... cut ...]
Apr  2 13:59:32 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/e75e2a2e/start.elf not found
Apr  2 13:59:32 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/autoboot.txt not found
Apr  2 13:59:32 raspi-boot-srv-aaq dnsmasq-tftp[31943]: sent /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/config.txt to 10.20.35.130
Apr  2 13:59:32 raspi-boot-srv-aaq dnsmasq-tftp[31943]: file /var/lib/piserver/tftproot/b8-27-eb-5e-2a-2e/recovery.elf not found

whats wrong ?

maxnet, 2021-04-02:
Does it work any better if you stick a SD card with just bootcode.bin into the Pi?

m-gavrilyuk, 2021-04-05:
yes it help.
I just found than client is already been boot with usb flash drive 
seems it conflict with lan boot when usb is connect  
and my problem with link-local address https://github.com/raspberrypi/piserver/issues/105#issuecomment-623521732
steel available on new server maybe its because piserver is the virtual machine ( 1 in Vmware and second on Hyper-V )

closed 2021-05-07 by m-gavrilyuk