raspberrypi/piserver #126: PiServer Client Network Security

mark-iiot, 2021-04-21:
Hi, We have the PiServer running on a VM (vmware). The connected Rpi clients are basically just running a chromium web interface for a .Net framework solution. Its basically a shop floor data capture solution. All is running fine. The question i have is in terms of security at the client end. On the PiServer we have configured the auto updates, Fail2ban, and UFW firewall etc but only the Fail2Ban service will run on the client image. Any advice?

maxnet, 2021-04-21:
So what daemons are you running on the client that you need fail2ban for?

mark-iiot, 2021-04-22:
This configuration is running on a commercial domain network and i guess being a little bit new to linux and the use of Raspberry Pi's i am obviously a little bit concerned about security and the potential vulnerabilities that might be exposed using this type of setup, specifically from the Raspberry PiServer client side. We have created a separate VLAN for this industrial network but cant necessarily completely lock it down as it needs to communicate with the commercial side so the automated security updates and firewall was hopefully an additional measure to secure the gaps but cant seem to get them working on the PiServer client image? But any advice appreciated!

maxnet, 2021-04-22:
You mentioned you installed fail2ban on the client side.
Why exactly?
Because you read somewhere it would add security?

Or is there some server process running ON THE CLIENT that it is actually protecting?
fail2ban is software to block access to remote IPs, if those IPs tried to login multiple times unsuccessfully to a server process such as a SSH server, webinterface, etc.
If there is no server process on your client, it does not do anything.

I still have no clue against what perceived threat you are trying to protect the client.

Typical threats are more in the domain of rogue DHCP servers interfering with the boot process.
Something a proper managed L3 Ethernet switch can prevent. Either by enabling DHCP snooping, or putting each and every client in a separate VLAN and routing instead of switching on the switch.

maxnet, 2021-04-22:
To illustrate what I mean by managed Ethernet switch configuration.

If you have one of the better L3 Ethernet switches -like those made by Juniper- you can simply enable "DHCP snooping" in the VLAN your clients live:

[image]
"

And put a tick on "Trust DHCP" on the Ethernet port Piserver is connected to:

[image]

If Piserver is not handing out the IPs as standalone DHCP server, you also need to make such tick on the Ethernet port your main DHCP server is connected to.

If any other devices connected to your industrual VLAN then tries to play DHCP/PXE rogue server the switch will prevent that and filter the traffic out.

Can typically also see it in the switch logs, if that happens:

[image]

With less capable L3 switches that do not support DHCP snooping, you need a separate VLAN and subnet for each client.
So if you have 32 clients, you need 32 VLANs.
And you then configure DHCP relaying from those 32 VLANs to the Piserver IP.
A lot more work to setup, but that will also prevent rogue DHCP servers from being able to interfere/hijack the clients.

mark-iiot, 2021-04-22:
Thanks maxnet our switches do support DHCP snooping fortunately and very much appreciate your advise. The answer to your previous comment is yes pretty much exactly that! When querying the best practice approach to Raspberry Pi security the three main suggestions that regularly came up was fail2ban (brute force attacks), firewall (ufw for simplicity), and security updates in this case through the unattended_upgrades. Admittingly i had taken this literally but take your point on fail2ban on the client side. I guess i can sleep easy :) Many Thanks!

closed 2021-04-22 by mark-iiot