raspberrypi/piserver #129: feature request / import mac's for supported rpi's via csv

heini66, 2021-05-04:
atm i'll have to boot each rpi that i want to have in a new piserv config'ed, i'll hav to boot them all each once  to get the mac's from them an d adopt them into the new piserv. woild be nice to have an export/import option fot the mac's and the images thes are assigend to....

maxnet, 2021-05-05:
Not sure how common the use of this functionality would be.
I would imagine that for most users that want to migrate their Piserver instance to another server, preserving user information may be a higher priority, than not having to power cycle each Pi once...

You do can transfer the information manually (e.g. with rsync).

- Make sure Piserver is fully installed on the new server first (go through the first use wizard).
- Transfer:

/var/lib/piserver/hosts.json has the list of MAC addresses.
/var/lib/piserver/installed_distros.json has the list of installed operating systems.
Transferring the symlinks in /var/lib/piserver/tftproot preserves the mapping between MAC and operating system.
/var/lib/piserver/os has the operating system files

- Call "sudo piserver --update-ip" after transferring files to have dnsmasq DHCP server configuration regenerated.

Do NOT transfer settings.json to the new location, as it contains an internal password to access the local LDAP server that will be different on the new server. (Set to something random during installation)

heini66, 2021-05-05:
i know about /var/lib/piserver/hosts.json, the symlinks in /var/lib/piserver/tftproot, /var/lib/piserver/installed_distros.json and i'm also fine with copy files via scp.
 but: i don't kow if other user will be able to get rid of this stuff. i'm thinking piserver('s) in a environment with 1-2-3 hundert attched client rpi's and 1,2,3,n piserver ( i386 based (x64 stuff for this would be anther kick ) on real or virtual machines
 next point for me is to get the lpad server configable after the initual setup of the piserv. ( to make the ldap-server changeable on a runnig instance ) should i open another request for this ???

maxnet, 2021-05-06:
>but: i don't kow if other user will be able to get rid of this stuff. i'm thinking piserver('s) in a environment with 1-2-3 hundert attched 
>client rpi's and 1,2,3,n piserver 

Cannot have more than one Piserver in practice, if you want users to be able to have the same central home directory...

> ( to make the ldap-server changeable on a runnig instance ) should i open another request for this ???

Do not think that will be supported. To many caveats.
Problem is that if you change LDAP server the user id numbers will no longer be the same as before, and as a result the permissions of all your home directories wrong.