raspberrypi/piserver #130: Ping does not work for non-sudo users

jonwitts, 2021-05-25:
When customising our client OS, non-sudo users on the client Pis are unable to run ping. They get the error: "ping: socket: Operation not permitted"

Performing the following command in the CHROOT fixes the issue: 

chmod 4711 /bin/ping

XECDesign, 2021-05-25:
I've just booted Raspberry Pi OS through piserver and am not seeing the issue.

By default, when `iputils-ping` gets installed, it tries to run `setcap cap_net_raw+ep /bin/ping` and if that fails, it runs `chmod u+s /bin/ping`. If I recall correctly NFS, doesn't support capabilities, so if you install iputils-ping on ext4 and then try to boot it from NFS, you'll see this issue. When we build Raspberry Pi OS images, everything is wrapped through `capsh --drop=cap_setfcap` to prevent this situation.

jonwitts, 2021-05-25:
I will try and do a little more testing on this, as when I had the vanilla Raspberry Pi OS running I didn't see the issue... (I think!)..

But after I installed my additional software I did... I can't see what in my customisations would affect the ping utility though... I will try running through all the customisations again to see if I can pin down which one breaks ping...

maxnet, 2021-05-25:
Perhaps the TS did "apt upgrade" from the chroot shell?

Seems the package was updated recently: https://metadata.ftp-master.debian.org/changelogs//main/i/iputils/iputils_20180629-2+deb10u2_changelog

And if run "apt upgrade" on the server setting the capability may succeed, as the ext4 system ON THE SERVER does support them.
Just cannot be exported through nfs afterwards.

jonwitts, 2021-05-25:
I did indeed perform a full dist-upgrade as part of my customisation...

jonwitts, 2021-05-25:
I will try bringing a vanilla Raspberry Pi OS down; running ping and then performing a dist-upgrade this afternoon to see if that replicates the issue.

jonwitts, 2021-05-25:
OK, so the version of ping does not show as changed after an dist-update but doing so does break the permissions on ping. I am guessing that @maxnet has the reason...

maxnet, 2021-05-25:
>OK, so the version of ping does not show as changed

If you mean "ping -V" shows same version number, that is expected.
Under Debian policy it is kinda forbidden to offer real upgrades to newer versions through the repository.
Those are supposed to wait until a new Debian release comes out every 2 years.
Individual stability and security patches do can be backported to the current version offered by the package.

"apt-cache policy iputils-ping" do should show newer version after upgrade.

Not sure what the proper fix is.
Could probably change the "chroot shell" fuctionality to call "capsh" similar to what pi-gen is doing. And that will fix it for folks that use that to run "apt upgrade"
However that will not correctly solve it for other users that make their own custom SD card first, run "apt upgrade" on their SD card as part of that, and import their custom OS from SD card to Piserver after that.

jonwitts, 2021-05-26:
A fix for the CHROOT method would certainly be appreciated; I now do all of my customisation through that method as I have found a far nicer way to work with PiServer images than trying to import from customised SD Card images... 

Although reading the Raspberry Pi forums it seems many people are wanting to use the SD Card method...

commit d9719a82 (2021-05-27): Drop CAP_SETFCAP when running chroot shell

Closes #130
closed 2021-05-27 by maxnet