raspberrypi/piserver #131: New Pi 4 8Gb MAC address

jonwitts, 2021-05-27:
Hi there,

I am just trying to add a new Pi4 8GB to my Pi server and it appears as if the MAC address for this device does not match the OUI in the MAC filter on PiServer. It's MAC address is e4:5f:01:0e:ff:25

At the end of /etc//dnsmasq.conf I have found the following: 
```
    # Delays sending DHCPOFFER and proxydhcp replies for at least the specified number of seconds.
    dhcp-mac=set:client_is_a_pi,B8:27:EB:*:*:*
    dhcp-reply-delay=tag:client_is_a_pi,2
```

but I thought that there were already 2 different MAC ranges that PiServer could accept? 

Where would I need to add the new MAC range for my Pi4 so that it can be added to PiServer?

Thanks,
Jon

jonwitts, 2021-05-27:
I can see on the 0.7 pre-release this commit has added the new OUI filter to match this MAC address:
https://github.com/raspberrypi/piserver/commit/e14022c8510b417e60e95e0019489a3f219e783d

Is there any way I can add this to my existing installing without having to install a pre-release package on my production server?

Thanks,
Jon

XECDesign, 2021-05-27:
Apologies, I didn't see that there was a 0.7.1. I'll add it to the repo.

jonwitts, 2021-05-27:
> Apologies, I didn't see that there was a 0.7.1. I'll add it to the repo.

Great, does that mean I will get it through apt upgrade on my server?

maxnet, 2021-05-27:
>Apologies, I didn't see that there was a 0.7.1. I'll add it to the repo.

Wait half an hour while I check in 0.7.2

XECDesign, 2021-05-27:
Will do, thanks. Does `dnsmasq.conf` need anything new?

maxnet, 2021-05-27:
>Will do, thanks. Does dnsmasq.conf need anything new?

No
The dhcp-reply-delay workaround listed by the TS is only needed for the oldest generation network booting Pi 3.
Not for ones with newer MACs.

(And the stuff Piserver generates in /etc/dnsmasq.d, so TS was looking at wrong file)

maxnet, 2021-05-27:
>Will do, thanks.

Checked in the code.
But please test on your own system before checking into repo.

I am seeing issues on Pi 4/400 where the bootcode is unable to download start4.elf reliably on first try.

```
May 27 18:20:59 raspberry dnsmasq-tftp[600]: sent /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/config.txt to 192.168.178.228
May 27 18:20:59 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/pieeprom.sig not found
May 27 18:20:59 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/recover4.elf not found
May 27 18:20:59 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/recovery.elf not found
May 27 18:21:29 raspberry dnsmasq-tftp[600]: error 0 Early terminate received from 192.168.178.228
May 27 18:21:29 raspberry dnsmasq-tftp[600]: failed sending /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/start4.elf to 192.168.178.228
```

When the firmware retries the entire boot sequence a second time later it does work ok:

```
[...]
May 27 18:22:11 raspberry dnsmasq-tftp[600]: sent /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/config.txt to 192.168.178.228
May 27 18:22:11 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/pieeprom.sig not found
May 27 18:22:11 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/recover4.elf not found
May 27 18:22:11 raspberry dnsmasq-tftp[600]: file /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/recovery.elf not found
May 27 18:22:13 raspberry dnsmasq-tftp[600]: sent /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/start4.elf to 192.168.178.228
May 27 18:22:13 raspberry dnsmasq-tftp[600]: sent /var/lib/piserver/tftproot/b8-27-eb-5f-69-b0/fixup4.dat to 192.168.178.228
```

Not likely to be a problem in the piserver code, but more likely in firmware or with my network. But may want to test if you can reproduce just in case.

maxnet, 2021-05-27:
>I am seeing issues on Pi 4/400 where the bootcode is unable to download start4.elf reliably on first try.

Solved the problem I was seeing by upgrading the Pi 4 EEPROM firmware to pieeprom-2021-04-29.bin
So apparently already fixed.

(The old 2020-09-03 EEPROM firmware seemed to deal bad with situations there is packet loss and a tftp server sends a tftp packet twice.
Sending ACK 1478 in response to 1477 being sent twice.

[image]

Still not sure why I seem to have packet loss though)

jonwitts, 2021-06-03:
Hi guys,

Any update on when the new version of PiServer might make it into the APT repo? 

Thanks,
Jon

jonwitts, 2021-08-12:
Hi there,

I can confirm that I now have version 0.7.2 of PiServer through APT and that this now allows RPi with the new MAC address to be added.

Many thanks!

closed 2021-08-12 by jonwitts