raspberrypi/piserver #135: Trying to SSH in piserver client

marc-odp, 2021-09-23:
I am trying to activate SSH so that I could connect to piserver clients using SSH.
I checked and applied this : #34 
But when I do 'service ssh status' in chroot_image terminal, i get 
[FAIL]sshd is not running[FAIL]

when i do 'ssh localhost' in chroot_image it works though.
But when i do it remotely using the ip of the client, i get a 'connection refused'...

maxnet, 2021-09-23:
>I checked and applied this : #34

Any error message during that?

>But when I do 'service ssh status' in chroot_image terminal, i get [FAIL]sshd is not running[FAIL]

Correct. There will not be any services running in the chroot itself.
Still should be able to enable them there though (they will not run under the chroot then, but on the client Pi).

>when i do 'ssh localhost' in chroot_image it works though.

Because are ssh'ing to the SSH daemon running on the server (not in the chroot) in that case.

marc-odp, 2021-09-24:
No error doing #34 
`ssh-keygen -A` gives no message.
`systemctl enable ssh` gives this message :
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable ssh

>  you just configure them to be enabled on the client Pi

It seems that it is the part that I don't understand. 
Being under the chroot, how can I configure ssh to be enabled on the client Pi by default ?

maxnet, 2021-09-24:
>Being under the chroot, how can I configure ssh to be enabled on the client Pi by default ?

That is what `systemctl enable` is supposed to do.
`systemctl start` means "start right now", `enable` means "configure the system to run the service on startup"

systemctl normally does the enabling by making a symlink from `/etc/systemd/system/multi-user.target.wants/ssh.service` to `/lib/systemd/system/ssh.service` on the filesystem.

The chroot is the filesystem that is served to the client, when it boots. So on boot systemd running on the client should examine  /etc/systemd/system/multi-user.target.wants and start what it finds there (at least that is the simplified version of it).

Judging by the message there may also be a legacy sysv ssh init script present on your RPI OS, which may cause problems.
Could also be that the ssh regeneration stuff provided by RPI OS is causing problems (cannot write to the read-only root file system, and may be a prerequisite for executing ssh).
Does it work any better if you run `systemctl disable regenerate_ssh_host_keys` inside the chroot?

(Not near a Piserver installation right now, so cannot check myself)

marc-odp, 2021-09-30:
I typed `systemctl disable regenerate_ssh_host_keys` and now I can connect with SSH to the client. Thank you !
I am a little confused about this part though :
> enable means "configure the system to run the service on startup"

What is the "startup" moment ? PiServer is running all the time on the server.  Is the startup happening when the client load the system ? Or there is a need some how to "reboot" PiServer when changing system commands ?

When using `sudo raspi-config` inside chroot, it has an effect immediately (clients just need to reconnect) or some kind of Piserver reboot is needed ?

closed 2023-08-05 by marc-odp