raspberrypi/piserver #138: Can't get PiServer started on RPi4

MilesAheadToo, 2021-11-17:
I'm using an Argon One m.2 case for my RPi4, which I want to use for PiServer. I had in installed successfully and booting 4 RPI3b's. But I was struggled with the power. So I rebuilt by RPi4 using [usb-boot](https://forums.raspberrypi.com/viewtopic.php?f=29&t=196778), which is working fantastically once I gave up trying to create an img of the SSD I was using at first. I started with Rasp OS Lite and built from there.

The issue is that I now can't get PiServer to start. I followed all the instructions and it complied without error. But now I can't get it to start, and it doesn't give me any errors.

Thoughts?

maxnet, 2021-11-17:
>Well, not sure it is quite solved, but I plugged the monitor back into the RPi4 and started PiServer. It popped up right away.
>So the issue may be in displaying PiServer in XRDP? I was having the issue I reported above while logging in tot he RPi4 using 
>RDP from a Windows 10 laptop.

Has not been tested with XRDP

>But now I can't get it to start, and it doesn't give me any errors.

Double check it does not output any errors to console, by starting a terminal first, and then "sudo piserver"

locutusofsand, 2021-11-22:
I am actually currently accessing my piserver running on a RPi4 via XRDP with no issues. I am using Raspberry Pi OS on the server.