raspberrypi/piserver #140: Setting HOSTNAME for each RPi instance

MilesAheadToo, 2021-12-01:
It would be great if the hostname of each of the instances would be set to the Description. It is much easier then to connect to the instances using SSH or RDP. I will have as many as 20 RPis which I would like to name RPiv-n where v = board type [0-4] and n = the instance number.

Is there a way of setting the hostname after the system has booted? 

Well, put differently, I notice that I cannot login as 'pi' and the user names selected in PiServer are not added to the sudoers group. So perhaps a secondary request/question would be that the users added in PiServer can be added tot he sudoers group.

[image]

maxnet, 2021-12-01:
>Is there a way of setting the hostname after the system has booted?

Recall dhcpcd will fetch and use the hostname from DHCP if /etc/hostname is set to localhost
That of course does require that you are able to configure your DHCP server so that it hands out unique hostnames based on MAC.

Since most people use the DHCP relay functionality of piserver, and we are not the authoritative DHCP server, that is not something we can easily add to piserver.

>So perhaps a secondary request/question would be to include an option to add the users in PiServer to the sudoers group.

sudo is a solution to a problem.
What problem are you trying to solve?

Keep in mind that the root file system is read-only because it is shared by all clients.
No amount of sudo is going to let you write things to /

That said you can edit /etc/security/group.conf to add users to extra groups.
By default only people logging in to lightdm are added to extra groups like video and input.
Be aware that extending this to ssh logins adds security risk.
E.g. any student logging in through ssh can perform key logging on another student logged in to the Pi locally if you give ssh users access to the "input" group.
Naturally that is also the case if you were to give them access to sudo group.
So that is definitely not recommended in multi-user environments....

MilesAheadToo, 2021-12-01:
OK, understand better now. 

Sudo: I was hoping to run  raspi-config to set the hostname.

I don't know linux nearly well enough to know how to do it, but I presume the same command issued by raspi-config could be used to set the hostname to a value defined in PiServer. My suggestion was the description field, but perhaps it needs a new field?

I just trying to avoid setting static IP addresses for the Pi's. That is a lot of tedious work on the DHCP server.

maxnet, 2021-12-01:
>I just trying to avoid setting static IP addresses for the Pi's. That is a lot of tedious work on the DHCP server.

We are talking about adding 20 entries to the configuration?

I am afraid that does not justify custom development for a hostname configuration feature.

MilesAheadToo, 2021-12-01:
OK. I thought it would be useful to a lot of users.

closed 2021-12-10 by MilesAheadToo