raspberrypi/piserver #141: Using samba to access NAS

MilesAheadToo, 2021-12-08:
I'm trying to set up the PiServer clients to access a shared NAS of 6TB over several old drives. I can access the shares I have set up from multiple other computers. But I cannot access them from the PiServer clients. All the NAS drives are attached to the RPi4 running PiServer.

**Please Note:** I am looking for help to do this correctly rather than stating that there is an error with PiServer.

On the RPi4 I run the following:

```
sudo mkdir /mnt/3TB_HDD
sudo mkdir /mnt/1TB_HDD
sudo mkdir /mnt/700GB_HDD
sudo mkdir /mnt/1TB_SSD
sudo mkdir /mnt/100GB_SSD

sudo chmod 777 /mnt/3TB_HDD
sudo chmod 777 /mnt/1TB_HDD
sudo chmod 777 /mnt/700GB_HDD
sudo chmod 777 /mnt/1TB_SSD
sudo chmod 777 /mnt/100GB_SSD
```

I have the following info in **/etc/samba/smb.conf**:
```
[3TB_HDD]
comment = 3TB SATA HDD in USB3 Caddy
path = /mnt/3TB_HDD
writeable = yes
create mask = 0777
directory mask = 0777
public = yes

[1TB_HDD]
comment = 1TB SATA HDD in USB3 Caddy
path = /mnt/1TB_HDD
writeable = yes
create mask = 0777
directory mask = 0777
public = yes

[1TB_SSD]
comment = 1TB SATA NVMe SSD in USB3 case
path = /mnt/1TB_SSD
writeable = yes
create mask = 0777
directory mask = 0777
public = yes

[700GB_HDD]
comment = 700GB SATA HDD in case
path = /mnt/700GB_HDD
writeable = yes
create mask = 0777
directory mask = 0777
public = yes

[100GB_SSD]
comment = 100GB SATA SSD in USB3 case
path = /mnt/100GB_SSD
writeable = yes
create mask = 0777
directory mask = 0777
public = yes
```

I have the following in **/etc/fstab**:
```
UUID=FAECC471ECC429A9	/mnt/3TB_HDD	ntfs	defaults,noatime,nofail	0	0
UUID=1E7EB1147EB0E625	/mnt/1TB_HDD	ntfs	defaults,noatime,nofail	0	0
UUID=527672890C6EF7EC	/mnt/700GB_HDD	ntfs	defaults,noatime,nofail	0	0
UUID=3D7B599C2285E9EC	/mnt/1TB_SSD	        ntfs	defaults,noatime,nofail	0	0
UUID=2DF507A17967AEA2	/mnt/100GB_SSD	ntfs	defaults,noatime,nofail	0	0
```
I can access these drives without error on several machines.

On the PiServer clients I have installed **samba-common** and **smbclient** because I will not share any of the client folders. I only want to get access to the samba NAS drives.

On the PiServer, I selected Software and then an image that I can use successfully to login to one of the clients with an XFCE desktop. I then click on **Shell** so that I can adjust the image. This is where I run apt 

`install -y samba-common smbclient
`
I then run the following commands:

```
mkdir /mnt/3TB_HDD
mkdir /mnt/1TB_HDD
mkdir /mnt/700GB_HDD
mkdir /mnt/1TB_SSD
mkdir /mnt/100GB_SSD

chmod 777 /mnt/3TB_HDD
chmod 777 /mnt/1TB_HDD
chmod 777 /mnt/700GB_HDD
chmod 777 /mnt/1TB_SSD
chmod 777 /mnt/100GB_SSD

```
 I then run **nano /etc/fstab** and add these lines

```
//RPISERVER/3TB_HDD /mnt/3TB_HDD    ntfs	nofail,guest	0	0
//RPISERVER/1TB_HDD /mnt/1TB_HDD    ntfs	nofail,guest	0	0
//RPISERVER/700GB_HDD	/mnt/700GB_HDD  ntfs	nofail,guest	0	0
//RPISERVER/1TB_SSD /mnt/1TB_SSD	ntfs    nofail,guest	0	0
//RPISERVER/100GB_SSD   /mnt/100GB_SSD	ntfs	nofail,guest	0	0
```

I then set this revised image as the image for all the PiServer clients and reboot the clients.

Once I have logged into a client I can navigate over to the **File Manager** and all the NAS folders are empty, even though the total size is correct. Notice the size indicator at the bottom of the screen shot.

[image]

I checked the the latest version of **ntfs-3g** was installed at part of the samba-common and smbclient install, and I checked that the RPiServer is available.

```
tim@raspberrypi:~ $ sudo nmblookup -S __SAMBA__
192.168.86.231 __SAMBA__<00>
Looking up status of 192.168.86.231
	RPISERVER       <00> -         B <ACTIVE> 
	RPISERVER       <03> -         B <ACTIVE> 
	RPISERVER       <20> -         B <ACTIVE> 
	WORKGROUP       <00> - <GROUP> B <ACTIVE> 
	WORKGROUP       <1e> - <GROUP> B <ACTIVE> 

```

But I get these errors when I issue the command **sudo mount -a**:
```
tim@raspberrypi:~ $ sudo mount -a
[sudo] password for tim: 
ntfs-3g: Failed to access volume '//RPISERVER/3TB_HDD': No such file or directory

ntfs-3g 2017.3.23AR.3 integrated FUSE 28 - Third Generation NTFS Driver
		Configuration type 7, XATTRS are on, POSIX ACLS are on

Copyright (C) 2005-2007 Yura Pakhuchiy
Copyright (C) 2006-2009 Szabolcs Szakacsits
Copyright (C) 2007-2017 Jean-Pierre Andre
Copyright (C) 2009 Erik Larsson

Usage:    ntfs-3g [-o option[,...]] <device|image_file> <mount_point>

Options:  ro (read-only mount), windows_names, uid=, gid=,
          umask=, fmask=, dmask=, streams_interface=.
          Please see the details in the manual (type: man ntfs-3g).

Example: ntfs-3g /dev/sda1 /mnt/windows

News, support and information:  http://tuxera.com
ntfs-3g: Failed to access volume '//RPISERVER/1TB_HDD': No such file or directory

ntfs-3g 2017.3.23AR.3 integrated FUSE 28 - Third Generation NTFS Driver
		Configuration type 7, XATTRS are on, POSIX ACLS are on

Copyright (C) 2005-2007 Yura Pakhuchiy
Copyright (C) 2006-2009 Szabolcs Szakacsits
Copyright (C) 2007-2017 Jean-Pierre Andre
Copyright (C) 2009 Erik Larsson

Usage:    ntfs-3g [-o option[,...]] <device|image_file> <mount_point>

Options:  ro (read-only mount), windows_names, uid=, gid=,
          umask=, fmask=, dmask=, streams_interface=.
          Please see the details in the manual (type: man ntfs-3g).

Example: ntfs-3g /dev/sda1 /mnt/windows

News, support and information:  http://tuxera.com
ntfs-3g: Failed to access volume '//RPISERVER/700GB_HDD': No such file or directory

ntfs-3g 2017.3.23AR.3 integrated FUSE 28 - Third Generation NTFS Driver
		Configuration type 7, XATTRS are on, POSIX ACLS are on

Copyright (C) 2005-2007 Yura Pakhuchiy
Copyright (C) 2006-2009 Szabolcs Szakacsits
Copyright (C) 2007-2017 Jean-Pierre Andre
Copyright (C) 2009 Erik Larsson

Usage:    ntfs-3g [-o option[,...]] <device|image_file> <mount_point>

Options:  ro (read-only mount), windows_names, uid=, gid=,
          umask=, fmask=, dmask=, streams_interface=.
          Please see the details in the manual (type: man ntfs-3g).

Example: ntfs-3g /dev/sda1 /mnt/windows

News, support and information:  http://tuxera.com
ntfs-3g: Failed to access volume '//RPISERVER/1TB_SSD': No such file or directory

ntfs-3g 2017.3.23AR.3 integrated FUSE 28 - Third Generation NTFS Driver
		Configuration type 7, XATTRS are on, POSIX ACLS are on

Copyright (C) 2005-2007 Yura Pakhuchiy
Copyright (C) 2006-2009 Szabolcs Szakacsits
Copyright (C) 2007-2017 Jean-Pierre Andre
Copyright (C) 2009 Erik Larsson

Usage:    ntfs-3g [-o option[,...]] <device|image_file> <mount_point>

Options:  ro (read-only mount), windows_names, uid=, gid=,
          umask=, fmask=, dmask=, streams_interface=.
          Please see the details in the manual (type: man ntfs-3g).

Example: ntfs-3g /dev/sda1 /mnt/windows

News, support and information:  http://tuxera.com
ntfs-3g: Failed to access volume '//RPISERVER/100GB_SSD': No such file or directory

ntfs-3g 2017.3.23AR.3 integrated FUSE 28 - Third Generation NTFS Driver
		Configuration type 7, XATTRS are on, POSIX ACLS are on

Copyright (C) 2005-2007 Yura Pakhuchiy
Copyright (C) 2006-2009 Szabolcs Szakacsits
Copyright (C) 2007-2017 Jean-Pierre Andre
Copyright (C) 2009 Erik Larsson

Usage:    ntfs-3g [-o option[,...]] <device|image_file> <mount_point>

Options:  ro (read-only mount), windows_names, uid=, gid=,
          umask=, fmask=, dmask=, streams_interface=.
          Please see the details in the manual (type: man ntfs-3g).

Example: ntfs-3g /dev/sda1 /mnt/windows

News, support and information:  http://tuxera.com
```

maxnet, 2021-12-08:
>All the NAS drives are attached to the RPi4 running PiServer.

JBOD directly attached to the Piserver, without anything special like a seperate NAS device, right?

My first counter question would be, why Samba?
Do you have a need to access the files as well from Windows computers?

If not, you can also just add your mount points to /etc/exports on the Piserver box to have them shared through NFS instead, like we do for the root file system and the shared folders.
Or use SSHFS if you need security (which we use for the user's home folder).

>On the PiServer clients I have installed samba-common and smbclient because I will not share any of the client folders.

For just mounting CIFS (Samba) folders, you may want cifs-utils instead.

>I then run nano /etc/fstab and add these lines
>
>//RPISERVER/3TB_HDD /mnt/3TB_HDD    ntfs	nofail,guest	0	0

Samba does not speak ntfs, it speaks the cifs protocol.

If there is no security present, and you do NOT have a need to authenticate the user to the Samba server, you can try:

```
//piserver/3TB_HDD /mnt/3TB_HDD    cifs 	nofail,guest	0	0
```

("piserver" is inserted into /etc/hosts to point to the Piserver server IP automatically by our system)

If you do would like to authenticate the user to the server with the username and password the user entered when logging in, you can play with editing /etc/security/pam_mount.conf.xml

MilesAheadToo, 2021-12-08:
- USB3 hub connected directly to RPi4 running PiServer, no NAS
- Would be nice to connect Windows systems to exchange info/data
-  I'll explore /etc/exports, never hear of this before. thanks for the tip.
- CIFS/NTFS - Interesting because I can access the drives from Windows, but I will try CIFS instead
- No authentication needed. This is all private, running behind a firewall.
- //piserver vs //RPISERVER - thanks, will try that

maxnet, 2021-12-08:
>CIFS/NTFS - Interesting because I can access the drives from Windows, but I will try CIFS instead

Note that "cifs" goes in the /etc/fstab for the client.

You do can keep it "ntfs" in the /etc/fstab of the server, if that is what your drives are formatted in.
(Although I would reformat them as ext4 instead, unless you have existing data on them you want to keep).

MilesAheadToo, 2021-12-09:
Thanks very much for your help @maxnet. I can now see the files, and navigate into the folders, but it seems that I have RO access.

[image]

On the the RPi4 running PiServer I added the rows at the bottom to **/etc/exports**

```
# /etc/exports: the access control list for filesystems which may be exported
#		to NFS clients.  See exports(5).
#
# Example for NFSv2 and NFSv3:
# /srv/homes       hostname1(rw,sync,no_subtree_check) hostname2(ro,sync,no_subtree_check)
#
# Example for NFSv4:
# /srv/nfs4        gss/krb5i(rw,sync,fsid=0,crossmnt,no_subtree_check)
# /srv/nfs4/homes  gss/krb5i(rw,sync,no_subtree_check)
#
/var/lib/piserver/os *(ro,no_subtree_check,no_root_squash,fsid=1055)

/mnt/3TB_HDD *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
/mnt/1TB_HDD *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
/mnt/1TB_SSD *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
/mnt/700GB_HDD *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
/mnt/100GB_SSD *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
```

Within PiServer I edited the correct image and adjusted **/etc/fstab** as follows:

[image]

I'm guessing that I should either use samba or NFS, but not both, on the clients.

maxnet, 2021-12-09:
If using nfs you should have a line like this on the client:

[code]
piserver:/mnt/100GB_SSD /mnt/100GB_SSD   nfs    defaults,nolock,rw
[/code]

Do keep in mind that for writes to succeed, the username that is logged in on the client will also need write access to the folder you are trying to write to.
So make sure your (chmod) permissions are correct as well.
(And not sure if those work correctly when using Windows' ntfs as a file system on the disks. May need to move to ext4 if you want writes to work properly).

maxnet, 2021-12-09:
>I used chmod 777 on both the RPi4 and the PiServer image when creating the /mnt/drive_name folders.

If you are trying to create the file /mnt/1TB_HDD/GoogleDrive/PiServer/Share/copy.pdf -like in your screenshot-
the permissions of the folder "/mnt/1TB_HDD/GoogleDrive/PiServer/Share" matter.
Not what you set on the parent folder /mnt/1TB_HDD ...

maxnet, 2021-12-09:
If using samba (not NFS), I recall you can tell it to let all access occur to a specific username ( guest account = username ? )
If using a single user, easiest is to chown all folders to that user.  "chmod -R username /mnt/1TB_HDD"

Again, no idea if permissions work properly at all if using a Windows file system like NTFS.
May need to switch to ext4 first.

MilesAheadToo, 2021-12-09:
Many, many thanks @maxnet, I got it working with NFS. I decided to separate out the NAS accessible to Windows via samba from the rw share I want to create for the RPi's connected to PiServer. 

I have another 120GB USB HDD so I decided to use this exclusively for PiServer. It is mounted on the RPi4 running PiServer.
- /etc/fstab on RPi4
`
UUID=2f3676cb-a54b-46aa-9e88-e62d95a45bff   /mnt/PiShare ext4	defaults,noatime,nofail	0	0`
- /etc/exports on RPi4
`/mnt/PiShare *(rw,all_squash,insecure,async,no_subtree_check,anonuid=1000,anongid=1000)
`

I can navigate to `/mnt/PiShare` on the RPi4 and I have full access to copy files, delete files, create folders, delete folders.

[image]

- PiServer client image /etc/fstab
`piserver:/mnt/PiShare   /mnt/PiShare    nfs defaults,nofail,x-gvfs-show,x-gvfs-name=PiShare 0   0
`
I can add, edit, and delete files and folders in PiShare. 
Again, thank you for your help.

closed 2021-12-10 by MilesAheadToo