raspberrypi/piserver #142: Placing files in the shared folder

MilesAheadToo, 2021-12-10:
I'm trying to load files into the 'shared' folder to make them accessible to all the clients. The folder is ro to the clients so I can't login to client and load the files from the client. I tried to find the folder on the system running PiServer, but couldn't.

I am trying to install Microsoft Visual Studio Code on the PiServer image. I can download a .deb file from a website, but since I cannot run a web browser in the CLI I downloaded it on the host system. Now I need to get it to a location accessible to the PiServer CLI so that I can install it on the image.

I know I must be missing something.

maxnet, 2021-12-10:
>I tried to find the folder on the system running PiServer, but couldn't.

/var/lib/piserver/shared
Recall there is also a button in the GUI to open the file browser to there. 
May need to create a subfolder in the GUI first

>I know I must be missing something.

Recall VS code is in the repository.
Try: apt update && apt install code

marc-odp, 2023-08-05:
VS code is now in the last os image. No need to install it any more.