raspberrypi/piserver #143: Add support for Ubuntu Desktop

PortalMario, 2022-01-08:
I tried to build the Raspberry Pi Ubuntu Desktop image for piserver, using the convert.sh and the steps from: #55 

However, during:
`chroot work qemu-arm-static /bin/bash /chrooted_cmds`
the following error occurred: `Invalid ELF image for this architecture` 

Would it be possible to make the raspberry ubuntu desktop image usable/compatible for piserver boot?

maxnet, 2022-01-08:
>the following error occurred: Invalid ELF image for this architecture

Probably need to use a 64-bit Linux distribution on the server, and replace `qemu-arm-static` with `qemu-aarch64-static` in the script, if you want to use a 64-bit Linux distribution with Piserver.
However suspect you need more work if that 64-bit OS is not Pi OS though...

>Would it be possible to make the raspberry ubuntu desktop image usable/compatible for piserver boot?

Only if you know what you are doing...

I am afraid the scripts are not designed to convert arbitrary Linux distributions.
They were written to convert images for classroom use, with central authentication, and storing student home folder's on the server.
And that requires different settings for different Linux distributions...
If you are not familiar with what it takes to configure that on Ubuntu, I wouldn't recommend you trying.

PortalMario, 2022-01-09:
Thanks for the quick response and the great Project! I guess you are right, this is fine for me. Closing this now.

closed 2022-01-09 by PortalMario