raspberrypi/piserver #144: 64 bit OS?

locutusofsand, 2022-02-07:
Will the newly released 64 bit OS be compatible with piserver?

XECDesign, 2022-02-09:
It's on the todo list and should show up within a week or two.

maxnet, 2022-02-09:
>It's on the todo list and should show up within a week or two.

Do make sure you also modify the 3 scripts that call qemu-arm-static right now, to detect architecture being used and use qemu-aarch64-static if needed instead.

https://github.com/raspberrypi/piserver/search?q=qemu-arm-static

locutusofsand, 2022-04-08:
Good morning,
I just saw the OS update from yesterday with the quiet upgrade to 5.15. Will the 64 bit OS make it into piserver soon? I have to rebuild one of my images and would like to move it to the 64 bit version and I can wait a couple of weeks if it would happen by then.

Thanks!

XECDesign, 2022-04-19:
Ran into a few issues while testing the piserver archives for the last release, so they're delayed until those are fixed. It shouldn't take too long.

locutusofsand, 2022-05-03:
I saw some changes come in and the OS show up in the list in piserver so for the curiosity in me I installed it and everything loaded fine and it boots for me, but it faults on an upgrade. This is the result right after the download of an apt upgrade.

```
Fetched 12.1 MB in 6s (1923 kB/s)                                              
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault
E: Sub-process /usr/bin/apt-listchanges --apt || test $? -lt 10 returned an error code (1)
E: Failure running script /usr/bin/apt-listchanges --apt || test $? -lt 10
```

XECDesign, 2022-05-03:
What's your host OS? The version of qemu might be a be too old. What does `qemu-aarch64-static --version` say?

Or maybe piserver explicitly calls `qemu-arm-static`, which won't work for arm64.

locutusofsand, 2022-05-03:
`qemu-aarch64 version 3.1.0 (Debian 1:3.1+dfsg-8+deb10u8)`
I am using Raspberry Pi Desktop for PC and Mac and it is current on updates.

XECDesign, 2022-05-03:
That would be the problem then, I think.

It should resolve itself when we have a new desktop release based on Bullseye out, which is a fairly high priority.