raspberrypi/piserver #146: apt update in shell on x86_64 bullseye: At least one invalid signature was encountered.

toonemastic, 2022-02-14:
Hello,
I've an issue when doing "apt update" in the shell on a Raspberry Pi image when using Piserver on a x86_64 Raspberry Pi OS running under bullseye.
I don't have this issue doing the same thing with the same image on a x86_64 Raspberry Pi OS running under buster nor on a Raspberry Pi 4 running bullseye 64bits.

See here under screenshot of what happens:

1) On the Raspberry Pi OS Bullseye:
pi@server-x86:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@server-x86:~ $ arch
x86_64

Running using the Shell on Raspberry OS Pi Lite:
root@server-x86:/# arch 
armv7l
root@server-x86:/# apt update
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB] 
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB] 
Err:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  At least one invalid signature was encountered.
Err:1 http://archive.raspberrypi.org/debian bullseye InRelease
  At least one invalid signature was encountered+ 
Fetched 38.5 kB in ls (50.4 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. 
GPG error: http://raspbian.raspberrypi.org/raspbian bullseye InRelease: At least one invalid signature was encountered.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. 
GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: At least one invalid signature was encountered.
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  At least one invalid signature was encountered.
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/bullseye/lnRelease  At least one invalid signature was encountered. 
W: Some index files failed to download. They have been ignored, or old ones used instead.
root@server-x86:/#  

2) The same on Raspberry Pi OS Buster:
pi@raspberry:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@raspberry:~ $ arch
x86_64

Running using the Shell on Raspberry OS Pi Lite:
root@raspberry:/# arch 
armv7l
root@raspberry:/# apt update
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease 
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
7 packages can be upgraded. Run 'apt list --upgradable' to see them. 
root@raspberry:/#

Do you have an idea on how to resolve that issue?

Thanks for your collaboration.

Pierre

toonemastic, 2022-02-15:
Hello,

I have been further after I've found a workaround on https://forums.raspberrypi.com
I don't exactly what this option given to apt-get resolve the issue but it does : 
Instead of using "apt update", I use "apt -o APT::Sandbox::User=root update".

That don't explain why the issue with apt update (At least one invalid signature was encountered) appears on my Raspberry Pi Desktop upgraded to bullseye running on my PC.

Why do I need a bullseye version running on my P instead of a buster version of Raspberry PI Desktop?
It is because, a lot of packages have more recent version available and running with less issues. I think especially about Epoptes needed to manage our labo.

The bullseyes version was created on the PC from a running buster version following these steps:

Replace every occurrence of the word "buster" with "bullseye"
sudo nano /etc/apt/sources.list
sudo nano /etc/apt/sources.list.d/raspi.list

Verify and do modification to deb http://security.debian.org into
deb http://security.debian.org/debian-security/ bullseye-security main contrib non-free

sudo apt update
sudo apt install libgcc-8-dev gcc-8-base
sudo apt full-upgrade
Answer Y to restart services without asking
Answer Y to use new config files
After error for not find dependencies:
sudo apt --fix-broken install
sudo apt full-upgrade
sudo apt autoremove
sudo apt full-upgrade
sudo apt autoremove
sudo apt full-upgrade

Hope it's help.
I don't know if you want to keep this issue opened but a explanantion about the issue can be useful.

Regards,

Pierre

XECDesign, 2022-02-16:
Sounds like https://gitlab.com/qemu-project/qemu/-/issues/633

toonemastic, 2022-02-16:
Hello,

Thanks for your rapid look at this issue.
It is well that qemu issue.
This command solve the solution. It is not permanent, it need to be relaunch after a reboot of the host.
I see that the the fix for qemu is now merged. So the issue will be fixed when new package will be created.
Thanks for your help.
I close now this ticket.

Regards,

Pierre

closed 2022-02-16 by toonemastic

XECDesign, 2022-02-16:
The problem is that even when the fix makes it into a qemu release, debian are unlikely to pick it up for Bullseye.

You can add a file like `/etc/sysctl.d/20-qemu-workaround.conf` with the line `vm.mmap_min_addr=53248` to make the change stick.

XECDesign, 2022-02-16:
I should also add that there are a few other issues you might run into with qemu, depending on what version you use:

AFAIK, this is still an issue (to be fair, it's probably not up to qemu to fix it):
https://gitlab.com/qemu-project/qemu/-/issues/263

One workaround for that issue is to run an i686 build of qemu, which can then trigger this issue:
https://gitlab.com/qemu-project/qemu/-/issues/385

So unless piserver ships with a known-working qemu build or fetches it from somewhere, it's likely people will be hitting all kinds of issues.

toonemastic, 2022-02-16:
Add a new conf file in /etc/sysctl.d. So the settings is now active after a reboot.
It is perfect for me like that.
Concerning the other issues, I will take a closer look.

marc-odp, 2022-09-23:
Thank you. You saved my day. 
I had exactly the same problem with the Official Bullseye version of Raspberry PI Desktop released recently.

descartes, 2023-08-02:
This still appears to be an issue and whilst I found it using my Google-Fu, it would be better if it was reopened and/or put on the readme.

`sysctl vm.mmap_min_addr=53248` resolves the invalid signature issue.

Uncommenting en_US.UTF-8 in `/etc/locale.gen` and then running `locale-gen` to resolve the error messages about locale and various environment variables.

Once done updates proceed normally and the whole system is freaking-awesome for creating an workshop setup with just the software required, preventing students fiddling with 'other things'.