raspberrypi/piserver #148: Request: Allow all Pis with network boot enabled to join Piserver booting without adding them to server explicitly

JamesMatchett, 2022-04-02:
Hi there,

I was wondering if there was a way to allow all Raspberry Pis clients with network boot enabled and on the same local network as Piserver to connect and boot without having to be added explicitly to the server as a client 

This would save a large amount of time in setting up our monthly Raspberry Jams

Thank you! 

James M (Northern Ireland Raspberry Pi Jam)

maxnet, 2022-04-02:
May work if you create a /etc/dnsmasq.d/mysettingsfile with:

```
dhcp-mac=set:piserver,b8:27:eb:*:*:*
dhcp-mac=set:piserver,dc:a6:32:*:*:*
dhcp-mac=set:piserver,e4:5f:01:*:*:*
``` 

And symlink the main TFTP folder to the boot folder of the OS you want to use, among the lines of:

```
# Figure out what the folder of the OS image you want everyone to use is
ls -l /var/lib/piserver/os

sudo mv /var/lib/piserver/tftproot /var/lib/piserver/tftproot.backup
sudo ln -s /var/lib/piserver/os/NAME-OF-THE-OS-YOU-WANT-EVERYONE-TO-USE/boot  /var/lib/piserver
```

(Normally the tftproot folders contains a lot of symlinks, one for each Pi configured, to the OS that Pi should use. But it defaults to using the parent directory if no specific match for a Pi's MAC address is found. So can just symlink the tftproot folder itself).

This is all theoretically. I haven't verified it works.

JamesMatchett, 2022-04-02:
Thank you so much for suggesting this approach @maxnet, I'll definitely give it a go during our setup next week and let you know how well it works for us.

Best, James 👍