raspberrypi/piserver #154: LDAP port setting - Piserver install

marc-odp, 2022-08-30:
Hi, I try to install Piserver on a new system (lubuntu)... This time, I want to experiment with ldap.
I compiled the source code with no problem and I started the program. 
When I reach the dialog box 'select authentication method', I try to put the address of the ldap server like this : 192.168.1.167:16389
but I get an error : "Error initializing LDAP connecting : Bad parameter to an ldap routine".
Maybe it is not possible to specify a non-standard port ? (16389 instead of 389)
Or is there another way to do it ?
-m-

maxnet, 2022-08-30:
>I try to put the address of the ldap server like this : 192.168.1.167:16389

Recall that if there is a : in the server address, the code assumes you are entering a full URI instead.
Like: ldap://1.2.3.4 or ldaps://1.2.3.4

So you may want to try if: ldap://192.168.1.167:16389 works better.

marc-odp, 2022-08-31:
Putting a full url doesn't work better. I get the message : 
`Error logging in LDAP server 'ldap://192.168.1.167:16389' with user 'user': Cant' contact LDAP server`
I think specifying a port number is not possible for some reason. When I check the log of the LDAP server, I don't see any login attempt.
When I do 
`ldapsearch -x -H ldap://192.168.1.167:16389 -D "uid=user" -w "secret"`
It works and log is showing activity. So the ldap server is working fine.
I tried using Jxplorer and it works too.

maxnet, 2022-08-31:
>When I do
>ldapsearch -x -H ldap://192.168.1.167:16389 -D "uid=user" -w "secret"

Does it also work without the "-x"?
As we do SASL authentication, not plain-text simple authentication.

And you do are running the ldapsearch line on the Piserver box, to rule out any network/firewall issues?

marc-odp, 2022-08-31:
> Does it also work without the "-x"? As we do SASL authentication, not plain-text simple authentication.

yes it does.

> And you do are running the ldapsearch line on the Piserver box, to rule out any network/firewall issues?

piserver install, ldapsearch and Jxplorer running on the same box.

marc-odp, 2022-09-10:
I could not make it work with another port so I am using now 389.
The LDAP server is accessible but the PiServer dialog box is giving me another error when I click on "Next" :
_Cannot use erase() with null_

marc-odp, 2022-09-10:
Credentials are ok. If I put a wrong Bind user or password, I get "_Invalid credentials_". So this is not the problem

maxnet, 2022-09-10:
>Cannot use erase() with null

Doesn't sound like something LDAP related, but more like an error that may come from the JSON library that we use to parse the repository meta-data when operating on invalid files.

Is there any type of Internet filtering active?
Opening a browser and going to https://downloads.raspberrypi.org/piserver.json gives something that looks like json, and not a page from filter software?

marc-odp, 2022-09-10:
It is json.

[image]

maxnet, 2022-09-10:
Could also be something wrong with the json files that carry Piserver's configuration.
Shouldn't happen on an initial installation though.

Can try closing wizard -> "sudo rm /var/lib/piserver/settings.json" -> Running the wizard again.

marc-odp, 2022-09-10:
I didn't have a settings.json file in /var/lib/piserver
Only the 3 folders : os scripts tftproot
Running the wizard again, I get the same error.
Is it not the result of a problem parsing the response of the ldap server ?

maxnet, 2022-09-10:
Managed to reproduce.
This build should fix it: https://github.com/raspberrypi/piserver/releases/download/0.7.5/piserver_0.7.5_i386.deb

marc-odp, 2022-09-11:
Because of some dependency problem (I am using lubuntu), I could not install the deb file. So I used the source code and it was perfect. No more problem, the wizard finished the installation. Thank you !

Any possibility to show the piserver version in the app ? Maybe in the settings tab ? I think it would be nice to add, to know which version is running...

closed 2022-09-11 by marc-odp