raspberrypi/piserver #155: LDAP - change default home folder

marc-odp, 2022-09-16:
Hi,
I put in my LDAP a different home folder for my users.
Instead of _/home/userX_
it is : _/piserver-homes/userX_
The idea is to separate the users of the host and the users of piserver.
I created the _piserver-homes_ folder on the host.
When the user attempt the first login in piserver, there is no error but the login dialog box comes back again and again.
I don't see how to fix this. Should I create a _piserver-homes_ folder in the root console ? Should I edit _/etc/fstab_ in the root console ? Or should I create a _piserver-homes_ folder in /var/lib/piserver/os/<name of os>
Or maybe the whole thing is just a bad idea and I should stick with _home_ being _home_ ? :-)
Thanks.
marc

maxnet, 2022-09-16:
That is not going to work out-of-the-box.
The user's home folder is going to be mounted at /home/userX on the client after login.

If you really want things differently, you can edit /var/lib/piserver/os/name-of-os/etc/security/pam_mount.conf.xml to mount at a different location and /var/lib/piserver/os/name-of-os/etc/fstab to create an empty tmpfs mountpoint at somewhere different than /home
But I would not recommend it.

marc-odp, 2022-09-27:
What about keeping the name home but having it mounted from NFS source ?
I am trying to make it work. I change fstab on the host -> works
I changed fstab in the root shell -> works after doing : mount -a
However, when connect to a client inside the root shell using : ssh user@192.168.1.x it is not working. I see my line in fstab for home but home was not mounted...
Is it the right procedure or I should use autofs ?
Please note that I am using ldap server. As I understand, I have to create the directory of the user manually.

marc-odp, 2022-09-27:
Well. I tested and it seems it is automatic. Great !
Thank you for your great support.

closed 2022-09-27 by marc-odp

maxnet, 2022-09-27:
>192.168.1.x:/xxx /home nfs defaults,nolock,rw

You do realize that if you are using read-write NFS v3 for home folders, there is no authentication for those?

As in: any person that is able to connect their own computer/phone to your network, or that is able to sneak in a normal RPI OS SD card into one of the Pi clients, can access the personal files of every student/teacher...
That is usually "not desired" to say at least (and a potential legal problem in parts of the world with privacy laws).

marc-odp, 2022-09-27:
Well, there doesn't seem to be a solution to that problem. Is there any ?
Is using CIFS and SAMBA better ?

maxnet, 2022-09-27:
>Well, there doesn't seem to be a solution to that problem. Is there any ?

We are using sshfs by default.
With that, once a user logs into the client, a SSH connection is established to the server, authenticated with the username and password the user entered, and files are transferred over that.
This is taken care of by pam_mount ( /var/lib/piserver/os/name-of-os/etc/security/pam_mount.conf.xml )

So the user only has access to its own home folder, instead of exporting everything under /home to every computer in the network.

So our default is secure, and my first counter question would be, why do you want to use a different solution than the default?

Do you have a need to store the files on a different server than the Piserver server?

Other alternatives are indeed CIFS done properly with per-user authentication through pam_mount (not by exporting all of /home without authentication).
Or NFS v4 with per-user Kerberos authentication.
The latter does requires that every system in the network has correct date and time (Piserver does provide NTP service, but double check it is working correctly if you are using a non-standard setup), and that forward and reverse DNS records are created for all hosts.
That may be possible to arrange in your case, but it is not trivial enough that we can easily integrate that in Piserver (as the idea of Piserver is that it fits into any existing network, without changing anything).
That is why for simplicity sshfs was chosen over NFS with Kerberos.