raspberrypi/piserver #157: Error message in pi client's browser : Your clock is behind

marc-odp, 2022-11-20:
Hi,
It is the second time that we have a problem like this : in the pi clients we get in the browser the error "your clock is behind".
I checked the server os (Mint) and the date and time are correct. I also checked the Piserver Shell and date and time are also correct.
However, the pi client has a wrong date.
I tried to simply restart the server completely and after reboot I was surprised to see that it was not fixed. 
After doing `timedatectl status` in the pi client I got the correct date and time.
I don't understand why this is happening since NTP service seems to be active.

[image]
[image]

maxnet, 2022-11-20:
>I checked the server os (Mint) and the date and time are correct.

You do are certain you are running a NTP server such as ntpd there that does share the time with others on the network?
As opposed to a NTP client like systemd-timesyncd that does not?

The log of that ntp server program does not indicate errors at any point?

marc-odp, 2022-11-20:
I meant that the date of that computer was fine, not that it was running a NTP server. Actually, I didn't do anything special. I was assuming Piserver was taking care of dates by itself, maybe getting the date from the router or from a NTP server on the internet. I really don't know from where Piserver clients are getting the date...

maxnet, 2022-11-20:
>I was assuming Piserver was taking care of dates by itself

It takes care of that by telling the clients to get the time by NTP from the "piserver" server, as there is no guarantee Piserver clients have reliable Internet.

https://github.com/raspberrypi/piserver/blob/1404eabb1865ae4cfd88b3dfdfc5632714aa7368/scripts/convert_chrooted_cmds#L122

And it normally installs the ntp daemon out of the "ntp" Debian package on the server, during the first run wizard.
Since you have your own custom Linux distribution, and are not using RPI OS for Intel, it's hard to tell if that worked properly in your case, though.
Any other ntp software you may have on your Linux distribution that does not share time over the network, but does  claim the ntp port, can conflict.
Any firewall rules you may have, can also conflict.

marc-odp, 2023-07-28:
Indeed I was not running ntpd in Mint. After installing packages, starting the service rebooting, it worked.

closed 2023-07-28 by marc-odp