raspberrypi/piserver #160: No audio

marc-odp, 2023-07-13:
Hi,
I have speakers in all my monitors but I get no audio. So I thought that it is because hdmi audio is not the default option in PiServer.
I started raspi-config in Piserver's Shell but when I want to see the sound options I get the message 'No internal audio devices found'.
How can I do the setup for the sound ?
Thanks.
marc
[image]

maxnet, 2023-07-16:
Does it work any better if you try raspi-config's non-interactive mode?

`sudo raspi-config nonint do_audio 0`
or
`sudo raspi-config nonint do_audio 1`

marc-odp, 2023-07-25:
Thank you.
I noticed that right clicking on the speaker icon on the right corner of the screen in the client brings a menu where it is possible to chose hdmi, so we are using that system for the moment and we get the sound. The option is memorized in the user profile it seems so no need to do it again and again.
I will try your solution when I get rid of my other problem (I create another topic).

closed 2023-07-25 by marc-odp