raspberrypi/piserver #166: Piserver and Pi5 - Not an issue, just a question

marc-odp, 2024-02-15:
Did anybody try Pi5 with Piserver ? 
Is it working ?

jonwitts, 2024-04-03:
I haven't tried it yet but it's on my list of things to do...

jonwitts, 2024-04-09:
First hurdle is going to be getting Raspberry Pi OS Bookworm running on the server... 
I've created a SD Card from the 64bit Bookworm image and tried to add it to my PiServer - but after about half an hour I just get an "Error running convert script" message...

Any clues as to how we can get Raspberry Pi OS Bookworm running from PiServer @maxnet ?

jonwitts, 2024-04-10:
As well as needing to get Bookworm running on PiServer the list of allowed MAC OUIs will need updating in PiServer.

Currently on RPi devices with MACs starting b8:27:eb | dc:a6:32 | e4:5f:01 | d8:3a:dd can connect to PiServer. My Pi5 has a MAC address which starts 2c:cf:67

marc-odp, 2024-06-22:
@jonwitts : Hi Jon, I saw your posts on https://forums.raspberrypi.com/viewtopic.php?p=2220399&hilit=piserver#p2220399
It seems that Raspberry Pi Foundation has dropped total interest in Piserver.
We have two computer labs with raspberry pi 3 and 4 in use. My plan is to continue to use them as long as possible. Just like I did with XO laptops. However, for our new lab to setup this summer, I plan to use 'FOG Project' image management with LDAP and NFS shares.
As far as I am concerned, Raspberry Pi is part of history. Not only there is no evolution of the software but nowadays you can find cheaper and more powerful systems using NUC with N100 cpu.
-marc-