raspberrypi/piserver #167: Piserver's future - Not an issue, just a question

marc-odp, 2024-02-22:
Hi,
We are currently running a Piserver lab in a school in India with 50 Pi4. Though I had some difficulties in the past, the setup is now running well. I am using Poe switches with Poe splitters. Piserver is in a VM on a NAS and we have a ldap server in a docker container.
When we started adding devices to the Piserver lab (back with Pi2, Pi3,...) the price of Pi computers was cheap compared to the NUC type of computers. It is not the case anymore if we consider the Pi 5...
Soon, I will have to take the decision to create a new lab for another school and I wonder if I should follow the piserver path or not...
I see less activity here and I wonder if it is because nobody has problems or because nobody is using piserver any more.
Is piserver still a good option for setting up new lab projects ? That is the question.
-m-

maxnet, 2024-02-24:
I see two separate issues in this.
The question if anybody is using Piserver anymore.
"anymore" implies it did was used before.
Hoewever it has never been promoted by RPI, other than by a single blog post, decades ago.
So that it doesn't have much of a user base is to be expected.

And the question whether Pi are a good option as clients.
Despite the name (and the OUI_FILTER define in config.h), there isn't much in Piserver that requires Pi to be used.

XECDesign, 2024-02-26:
> Is piserver still a good option for setting up new lab projects ? That is the question.

I'd advise against it right now.

marc-odp, 2024-02-28:
I am curious... What would you advise to use ? Something like Veyon ?

> I'd advise against it right now.

XECDesign, 2024-03-04:
> What would you advise to use ?

Sorry, I don't have an informed answer to give. I'm not familiar with what options exist and what the real world requirements for this use case would be.

closed 2024-03-04 by ghollingworth